ArduPilot/ardupilot #3037: Trigger new flight mode "emergency land" on long failsafe action FS_LONG_ACTN

severinleuenberger, 2015-10-19:
this isssue is related to: 
https://github.com/diydrones/ardupilot/issues/2931
and concerns the same governmental requirement to bring an airplane down safely. Parachute ist indeed the preferred method to bring an emergency landing. But on a lot of fixed wing aircrafts there is no possibility to mount a parachute. 

Another method to get an aircraft somehow "safely out of the air" would be to trigger a generic "land mode". 

Compared to the normal "autoland mode" where you plan your autolanding trough mission planner, the "emergency land mode" should do the following
- trigger "flaps down" according LAND_FLAP_PERCNT
- set a sink rate to a specific value defined on TECS_LAND_EMERGENCY
regarding the sink rate, we reccomend 4m  per second as an acceptable default value. 
Smaller values let the plane fly to much away. Bigger values may cause to much kinetic energy on impact with risk to the ground collision objects, and more damage to the airplane.

Naterater, 2019-01-07:
I believe this has been implemented as "GLIDE" in http://ardupilot.org/plane/docs/parameters.html#fs-long-actn-long-failsafe-action. I would assume that the airplane adjusts its sink rate to maintain it's target airspeed TRIM_ARSPD_CM.

WickedShell, 2019-01-07:
`GLIDE` isn't the same thing at all, it doesn't account for flaps, landing gear, sink rates appropriately, and can go quite a long distance.

peterbarker, 2021-07-30:
Marking this for DevCall to make sure we'd accept a patch to add such a flight mode.

peterbarker, 2021-09-16:
We discussed this at the DevCall.  We thought the concept good, but also that it requires a chunk of thought - and really, an implementation, to discuss.

General thought was that it could reconfigure the aircraft for landing (flaps/gear/speedbrakes) and loiter down.  Potentially flaring if there's a rangefinder on-board.

Just as interesting was the concept of a `DO_FAILSAFE_START` mission item which would allow someone to create a min-mission to handle various failsafes in a semi-scripted manner, which might allow for implementing the above as well as some other options people have been looking at (loiter-down followed by qland)