ArduPilot/ardupilot #3038: If source is checked out only one level deep build fails

anathan, 2015-10-19:
Build Environment: Windows 10 x64 - PX4 Console (Msys)

Repro: Checkout (master) to /c/repo/ardupilot
cd /c/repo/ardupilot/Arducopter
make px4-v2

observe: Make/Sh.exe terminate and dump stacktrace after errors printed about finding nuttx config tool.

Take the exact same repo, move it to /c/repo/test/ardupilot and run the same commands, builds fine.

rmackay9, 2015-10-19:
Not sure what the issue is but it could be that the directory capitalisation is incorrect.
Perhaps it should be ArduCopter instead of Arducopter.

magicrub, 2015-10-19:
If you're playing musical directories you should do a clean.

WickedShell, 2018-03-10:
This is a support issue, and best handled on https://discuss.ardupilot.org (although I believe the capitalization issue probably is the cause)

closed 2018-03-10 by WickedShell