ArduPilot/ardupilot #3044: ArduCopter HIL firmware compiling not working

ComfyMobile, 2015-10-20:
Can I build ArduCopter HIL mode firmware? When trying to compile the current release we get error about variable "sonar" absence in Copter.h. Thank you!

magicrub, 2015-10-20:
In Copter 3.3 and Plane 3.4 you no longer need to compile for it. It's a param.

closed 2015-10-20 by magicrub

ComfyMobile, 2015-10-20:
how can i switch on this param? where? Thank you very much for help!

magicrub, 2015-10-20:
My mistake, the param HIL_MODE is only on plane. I thought it was global.
http://plane.ardupilot.com/wiki/arduplane-parameters/#hil_mode_enable_arduplanehil_mode

-TomP

rmackay9, 2015-10-21:
@ComfyMobile, @magicrub, HIL in copter hasn't worked in literally years so I can't say I'm surprised it's not compiling or working.  I think it'll be some work to get it sorted.. but that's not something that I can personally take on in the near future.