ArduPilot/ardupilot #3052: Copter: add set-mode reason code

rmackay9, 2015-10-21:
Each caller to set-mode should provide a reason code.  These reasons would include:
1. set by pilot via RC
2. set by pilot via GCS
3. set by battery failsafe
4. set by gps failsafe

This would allow us some additional complexity in failsafe logic such as checking whether the user has overriden/cancelled the failsafe by changing modes.

R-Lefebvre, 2015-10-21:
Great idea.

R-Lefebvre, 2015-10-21:
Would the reason code be stored permanently, so we can always make intelligent decisions in the future based on who/why the vehicle is in the current mode?

I wonder if current mode, and current reason, should be stored as arrays that keep on being bumped back, so we could maintain a history of modes and reasons, and really make intelligent decisions? (ie: we have a battery failsafe event, but every time we do the user switches to manual control, we should stop doing failsafes.)  Something like that.

rmackay9, 2016-05-24:
this is in master now.

closed 2016-05-24 by rmackay9