ArduPilot/ardupilot #3059: SITL: sim_vehicle -I flag doesn't appear to be working

gmorph, 2015-10-22:
Looks like we can't run multiple instances of the simulation.  Probably a simple bug.

http://ardupilot.com/forum/posting.php?mode=reply&f=81&t=14080

Thanks, Grant.

magicrub, 2015-12-12:
I thought this was working, didn't @WickedShell talking about using it recently?

WickedShell, 2015-12-12:
I had a problem with location where the L was case sensitive and I was
using the wrong case....

peterbarker, 2016-01-05:
This is working no problems for me.  Just tested two ArduPlanes / one ArduPlane+one ArduCopter.

fbalaban, 2016-01-19:
Well, @gmorph said he managed to replicate it.. I'm not sure how @peterbarker you managed to make it work, but are you definitely sure it's the same case? I mean, have you tried to send two different waypoint files to the two instances and then check if they still have different paths? Maybe it does it on the second run? Btw, your two SITL, are by the same build? Maybe you are using two different clones?

Jesse-Millwood, 2016-11-07:
This works just fine for me. 
Once you start the instances, I found it a good check to look at what the output is just to make sure they are using two different udp ports (which increment with the instance switch)

OXINARF, 2016-11-07:
We have since changed to a Python script so it's possible it is now fixed.

@gmorph @fbalaban Are you still seeing this problem?

fbalaban, 2016-11-07:
Hi,
sorry but I had found a workaround and didn't follow the forum discussion which by the way has a changed link: http://discuss.ardupilot.org/t/sitl-multiple-instances-waypoint-files-are-broadcasted/6938/6.
I'm terribly busy, I cannot replicate it now or check if it's there. Please give me 3 or 4 weeks to get back to you. 

Thanks

OXINARF, 2016-11-07:
From that link it looks like the issue is in MAVLink and MAVProxy, not in ArduPilot.

@peterbarker You commented in that thread, can you confirm the problem isn't in ArduPilot?

Jesse-Millwood, 2016-11-07:
Sorry I misunderstood the problem since the original link wasn't going to the correct place. A while ago I had an issue where the port numbers weren't changing when you had multiple instances. This is is what I thought the issue was. But I do have this issue with the waypoint file being broadcast and re-writing the wp file for the other instances.

peterbarker, 2016-11-10:
@OXINARF I do not believe there to be any issues with SITL's -I parameter; the ardupilot binaries AFAIK increment their listen ports appropriately.

OXINARF, 2016-11-10:
Alright, this issue looks like to be linked but external to ArduPilot so I'll close this.

closed 2016-11-10 by OXINARF

hatyar, 2017-12-07:
so this is what I try:

dronekit-sitl copter-3.3 --home=32.1542,34.83246,54,0 --instance=2

And this is what I get:

bind failed on port 5760 - Address already in use

I thought the address it would bind to is: 5780!

khancyr, 2017-12-07:
you need at least copter 3.5 or master

hatyar, 2017-12-07:
Thanks!

What is master?

magicrub, 2017-12-11:
>What is master?

"Master" branch in git. It's the main development branch.