ArduPilot/ardupilot #3064: AP 3.4: flat turns with rudder under CRUISE do not seem sticky

marcmerlin, 2015-10-22:
Cruise works fine.
Turning with ailerons gives a new heading, and cruise sticks to it.
If I turn with rudder, the turn is very slow, but that's normal because it's a 2.6m motor glider. Turn is flat as advertised.
However, as soon as I release the rudder input, the heading springs back away from where I was pointing.
I can't tell if it goes all the way back to before the rudder turn, or not, but it definitely goes back some.

I don't have a clean minimalistic dataflash log showing this, but before I go get one, can you confirm whether this is indeed not how things are supposed to work, and whether you'd like me to tweak some settings first?
You can get my plane settings here: http://marc.merlins.org/tmp/4.BIN

OXINARF, 2017-06-04:
Is this still relevant?

marcmerlin, 2017-06-04:
will check on my next set of flights.

marcmerlin, 2017-06-06:
Ok, I checked on a sky eye (2m pusher), and rudder turns kind of worked, but as soon as I released the rudder, the plane would bounce back some amount of the turn that was just made.
On a BFG2600 (2.6m glider), it's the same thing, except worse, and strangely the rudder turn worked worse in one direction than the other

Let's have a look at https://youtu.be/n4iWlvrRA5E?t=6m50s
In mavexplorer, CRUISE starts at 14:27:45, it's the CRUISE 71s 

See how
1) roll isn't as flat at desired, especially on the 2nd rudder turn (the first one was flat-ish except at the start
2) see how roll bounces both times when rudder is removed, and when that happens, the heading regresses from the one I stopped the turn at.

[image]

15 minutes ago
http://marc.merlins.org/tmp/00000033_BFG_CRUISE_OSCILLATION.BIN

marcmerlin, 2017-06-08:
@OXINARF to be honest, this is not a super important feature for me, but do rudder turns work well for others with conventional aircraft? (not flying wings like @WickedShell uses :) ). If it's not really tested/used much, it's ok if it doesn't work that well, or at all. I was quite surprised that one rudder turn direction caused that much roll though, while the other one mostly gave a flat-ish turn (blue line in the graph above)

magicrub, 2017-06-08:
@WickedShell doesn't use flying wing, I do. Maybe you got us mixed up! :) He [mostly uses this funny plane](https://srp.aero/lynx/) that doesn't have ailerons which is quite the opposite to a flying wing.

marcmerlin, 2017-06-08:
Sorry @WickedShell lack of sleep and got you mixed up with @magicrub . I meant you Tom,sorry :)  And yes, rudder only planes that rely on dihedral only are not real planes in my book ;)

marcmerlin, 2017-06-09:
So, I just tried this on my Bixler2 that had just been tuned, and it was worse, doing a rudder turn caused the aircraft to pitch sideways about as much as if I had done a regular aileron turn.
Then I went back to my BFG2600 after retuning it, and it didn't work well either, and not the same whether I tried to turn left or right (just like in the example I gave above already)

WickedShell, 2017-11-05:
Alright I tracked this down today, your first log is no longer available, and your second one is apparently corrupt (the mavproxy tools can't make heads or tails of it).

So whats happening is unless stick mixing is enabled then `rudder_input` is set to 0. https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/radio.cpp#L233 inside plane we only unlock the heading track if there is a roll command or `rudder_input` is non zero. https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/ArduPlane.cpp#L723

This is straightforward to always check the rudder input regardess of stick mixing in cruise which would resolve the bug your seeing here, but would be a slight behavior change. Based on the comment on the code though I think it was the intended behavior to honor rudder inputs as changing the heading lock. 

@marcmerlin @magicrub @tridge is everyone fine with allowing rudder input to change the cruise lock direction regardless of stick mixing? I can say that for rudder only aircraft this change would have 0 impact on cruise. Tagged as a dev cal topic.

magicrub, 2017-11-06:
I don't use CRUISE mode, ever, but I would have assumed that the rudder would change your desired heading. Why would we not want that?

iskess, 2017-11-06:
I have an old request for similar behavior in auto mode. The option to use rudder to correct small track errors, to reduce conststant roll changes.  
#3729

WickedShell, 2017-11-06:
@magicrub I think we do, I asked it as a question here first because it would be a behavior change from current use, but I think it's the correct one.

marcmerlin, 2017-11-06:
@WickedShell  thanks for having a look and sorry, yeah, I deleted that log from 2015 in the original post
No idea why the 2nd one is corrupted. 
As for your suggestion, I'm not knowledgeable of all the things involved in that changed, but on paper it sounds good.

magicrub, 2017-11-07:
+1 for changing the behavior. What are your thoughts @tridge

WickedShell, 2017-11-07:
@magicrub conclusion from call is to definitely change it, I'll send in a PR tonight for it.

closed 2017-11-08 by tridge

magicrub, 2017-11-08:
Implemented and merged, thank you @WickedShell !!

marcmerlin, 2017-11-09:
indeed, thank you!

PR #7195 (2017-11-08): Plane: Support unlocking cruise heading with rudder
This closes #3064 and allows the rudder to unlock cruise heading on plane.