ArduPilot/ardupilot #3070: Add support for generating CRC32 of flashed parameters

NaterGator, 2015-10-23:
I enabled this on the PX4 side [here](https://github.com/PX4/Firmware/pull/2982).

The premise is to generate a CRC32 hash of all parameter names and values that can be queried by a groundstation to validate a locally saved cache. The benefit is cutting out the time and bandwidth required to synchronize all parameters over the data link each time a new device connects. 

This CRC32 has is returned when the value for the "magic" parameter id _HASH_CHECK is requested. _HASH_CHECK should not be included as part of the  total parameter count or stored in EEPROM; it should simply be available as a reference point *if* the groundstation software chooses to request it (this keeps backwards and forwards compatibility with any groundstation software that does not implement a rapid hash-checked cache lookup on connect). 

I will be happy to have a go at implementing this myself if someone doesn't want to jump on it, with the caveat that it will be my first foray into ardupilot.

WickedShell, 2015-10-24:
+1 it's something I'm definitely in favor of. (Also ideally including
mission items at some point)

sharvashish, 2017-02-17:
Hi guys,

Any update on this issue?

magicrub, 2017-02-17:
no, it is still in high demand though,

DonLakeFlyer, 2018-06-21:
QGC code which generates CRC on gcs side is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/ParameterManager.cc#L825

DonLakeFlyer, 2018-06-21:
The protocol is:
* GCS sends PARAM_REQUEST_LIST
* Vehicle sends _HASH_CHECK as first param
* GCS checks against it's down cache
* If they match, GCS sends _HASH_CHECK value back in a write to get vehicle to stop streaming params

Parameters which are marked volatile in meta data do not take part in crc.

Regarding GND_TEMP usage:
* This could be handled with another meta data marking which would indicate volatile but don't write values back to cache.
* Or it could be changed to a command

DonLakeFlyer, 2018-06-21:
The other point is that this would allow you to back off a bit on the use of *_ENABLE parameters hiding parameters which would make QGC implementation much easier.

DonLakeFlyer, 2018-06-21:
The current definition of a volatile parameter is:
* A parameter which changes on the vehicle side, but the new values it not sent to the groundstation at the time of change
* A parameter which is different at vehicle boot from the last vehicle connection

MikeCharikov, 2019-10-23:
Maybe we can just add parameter to Ardupilot like PARAM_HASH and allow GCS to fill it, so GCS could read it later and check it with own CRC. 

Different GCS may have different lists of parameters they want to check, so may be it is not the good idea to bring CRC-logic onboard.

peterbarker, 2021-07-30:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/3685

closed 2021-07-30 by peterbarker