ArduPilot/ardupilot #3074: No vehicle state after starting Vagrant SITL instance

franknitty69, 2015-10-24:
after starting a vagrant SITL instance, there is no vehicle state. simulated vehicle doesn't appear to be running. previous versions of vagrant worked correctly, this issue is for repositories cloned after 2015/10/22 (maybe earlier). also the same issue occurs on windows and osx.

**steps to reproduce**
1. clone repository
2. install virtualbox
3. install vagrant
4. cd to arudpilot/Tools/vagrant
5. issue vagrant up command
6. after vm is running, issue vagrant ssh -c "sim_vehicle.sh -j 2" command
7. following messages are displayed:
msg: building sitl ...
msg: starting up at CMAC
msg: starting ardupilot ...
msg: window access not found logging to /tmp/ardupilot.log
msg: connect tcp:127.0.0.1:5760 source_system=255
msg: log directory: 
msg: telemetry log: mav.tlog
msg: waiting for heartbeat from tcp:127.0.0.1:5760

**result**
MAV> prompt is displayed. no vehicle state information is displayed and commands such as output does nothing except creates another MAV>prompt

**expected result**
vehicle state is displayed and mav commands work

a7scenario, 2015-11-02:
+1.  I also never get past "msg: waiting for heartbeat from tcp:127.0.0.1:5760".  

I'm using the latest ardupilot 3.4dev code from github, OSX El Capitan, and latest released pymavlink.

magicrub, 2015-11-02:
I've never gotten Vagrant to work. Tried twice on two separate clean
installs.

sigison, 2015-11-02:
I have the same problem, and that's without using vagrant.

danielerecchia, 2015-11-03:
I have the same problem.
I use cygwin e mavproxy.
Who tell me how solve it? thanks.

danielerecchia, 2015-11-03:
@franknitty69 have you solved it?

a7scenario, 2015-11-03:
The Vagrant VM SITL definitely works on El Capitan OSX using the wiki setup guide, but "breaks" for me sometime after 30 July 2015 version of Ardupilot code.  I suggest first trying to setup the SITL using pre-30 July 2015 Ardupilot code and see if the "waiting for heartbeat from tcp:127.0.0.1:5760" issue resolves.

franknitty69, 2015-11-03:
as of 2015/11/03, if i clone **diydrones/ardupilot** and then vagrant up from, **arudpilot/Tools/vagrant**, i get the same issue.

this vagrant sitl instance is working for me. its from Dronekit.
vagrant init 3drobotics/ardupilot-sitl

so as of now, i am using the vagrant instance referenced on the Dronekit website.
http://python.dronekit.io/guide/sitl_setup.html

lokinell, 2015-11-14:
I suffer from the same issue :(

andersonpaac, 2015-11-16:
I'm running it on a raspberry pi and I've the same issue!

derekxm, 2015-11-16:
I'm running on ubuntu 14.04 and same issue. No heartbeat from sitl.

derekxm, 2015-11-17:
Found that sitl is getting stuck in 
lin 291 of libraries/AP_HAL_SITL/UARTDriver.cpp
_fd = accept(_listen_fd, NULL, NULL);

krisgry, 2015-11-30:
Are there any plans on including this fix into master? A pull request from @derekxm maybe?

peterbarker, 2016-01-05:
Thanks for this report!

Commit 2c78b4f537241f6e3eb1d8f911d8fcb71034433f should have fixed this problem for you - now in master.

If problems persist, ... please reopen this issue.

closed 2016-01-05 by peterbarker