ArduPilot/ardupilot #3076: Copter: Flip mode glitch

earsneyes, 2015-10-24:
I stumbled upon a problem with flip mode in AC 3.3 / 3.4-dev.

When you trigger a flip it will roll (to the right) after doing the flip.

Sitl does it as well:
	
Start SITL:
cd ~/ardupilot/ArduCopter
`~/ardupilot/Tools/autotest/sim_vehicle.sh -v ArduCopter -L HAM --map --console`
`arm throttle`
`rc 3 1666`
`flip`

Got it on video for the thrill: https://www.youtube.com/watch?v=FoJkj9W1mCI

Log: http://firmware.bbbmini.com/24102015_flipmode.zip

imrj, 2015-10-25:
I have duplicated the same issue over and over again with AC3.3....abrupt turn to right after flip

magicrub, 2015-10-25:
smells like an integrator that needs be gated during the flip

R-Lefebvre, 2015-10-26:
At what point to you change back out of Flip mode?  (ie: Does the
roll-right error stop by itself eventually, or do you have to take back
manual control?)

earsneyes, 2015-10-26:
Flew in stabilize then triggered the Flip with Channel 8 (spring loaded switch)

The roll stops by itself after the short roll to the right. 
I just needed to counter steer to get it back on the ground.

rmackay9, 2015-10-27:
I've reproduced in the simulator so it's certainly a code problem.  I'm looking into it'll be a simple or more difficult fix.

earsneyes, 2015-10-27:
Thanks for looking into this.

rmackay9, 2015-12-10:
This is fixed in master now and will go out with Copter-3.4.  We're planning to start testing 3.4 in early January with the hopes of going live with it by end of Feb.

With a little more work we could probably come up with a patch to fix Copter-3.3 as well.  It should be a one line fix to update the earth-frame rates.  Obviously we'd like to patch it.. it's just a matter of time.

closed 2015-12-10 by rmackay9

earsneyes, 2015-12-10:
Perfect. 
Im going to fly with master soon. 
Thanks!