ArduPilot/ardupilot #3077: Garbled log file, EK2 failure

dave-dave, 2015-10-24:
Running ArduPlane master at commit ... not sure - a few days ago. I could read the log it would be easier.
I had a lot of compass arming fails, ran calibration, then it armed. 
First time for that issue for this hardware.
Hardware is Pixhawk, uBlox GPS, analog airspeed, PWM lidar.

Auto takeoff worked well after an initial tip stall (my fault) and some evasive maneuvers to dodge trees, but there was a bit of oscillation in auto flight. In FBWA, the oscillation turned to very nearly a total loss of control. Switching to manual allowed a recovery and switching back to FBWA controlled flight presented no repeat of the problems.

Downloading the log file was onerous - 1/10th the normal speed (15.6kbps instead of 157kbps), and I found that it was not readable for MAVExplorer (Current) or APMplanner 2.0.19rc4.
I've pulled the SD card, and the downloaded file is correct, the SD card log and the downloaded log are identical.

 I'll add the log files to this when I can find somewhere to put them.

tridge, 2015-10-25:
hi Dave,
That log is full of zeros, no data at all. I suspect you have an issue with your microSD card. I'd try reformatting it.
Do you have the tlog for the flight?
Cheers, Tridge

dave-dave, 2015-10-25:
Weird, all the other files are good. I'll format it shortly.
All the related files are uploading now http://www.users.on.net/~beanilu/odd_ekf.(bin|tlog|tlog.raw|way.txt|parm)

tridge, 2015-10-25:
Looking at the tlog I can see that before the calibration the two compasses were not consistent. The compass consistency test has recently got stricter.
The tlog is losing a lot of data so it is hard to see why the tuning is off, but I can see that the power setup needs some work. The servo voltage was changing a lot over the flight.
There were a few points in the flight where the EKF2 attitude diverged from the EKF1, but without the dataflash log I can't analyse it further. 
If you fly again I'd suggest you set AHRS_EKF_TYPE=1, and then we can look at the DF log to compare EKF performance

dave-dave, 2015-10-25:
Can do, thanks.

rmackay9, 2015-10-26:
Although I'm perhaps overstepping my responsibilities here because it's a plane issue, I don't think we should be putting support issues into Github so I'm going to close this issue.
I suspect Tridge will still see comments posted here in any case.

closed 2015-10-26 by rmackay9

dave-dave, 2015-10-26:
I understand why it has been closed, but where would I put possible
problems in code under active development, if anywhere?
Thanks.

rmackay9, 2015-10-26:
@dave-dave, I think drones-discuss@googlegroups.com, gitter or APM:Forum.