ArduPilot/ardupilot #3083: SITL: allow for other aerodynamical models than Rascal

krisgry, 2015-10-26:
Hi,
I am involved in a research project where we are using a different aerodynamical model than the Rascal110 (the standard JSBSim model for ArduPlane). I had this working by (slightly) modifying the runsim.py startup script, but now I am looking to make this work with sim_vehicle.sh. I have a suggestion as to how I would do this, but I would like some feedback since there might be a long term plan/goal that I am not aware of.

1. Pass an argument to sim_vehicle that corresponds to the name of the new model (i.e. defaults to Rascal) which is passed on to ArduPlane.
2. In libraries/SITL/SIM_JSBSim.cpp the creation of the different startup xml files, all occurrences of "Rascal" is replaced with this argument.
3. Users can then put new JSBSim models in Tools/autotest/aircraft (with the same name as the argument) and they will be loaded.

At least in our case, the new JSBSim model is a separate git repository and considered highly work in progress. So I suggest that all subfolders in Tools/autotest/aircraft, except Rascal and arducopter, are ignored by git.

All this should be fairly straight forward to implement, but my concerns are with the fact that there already is a "model" parameter that is passed from sim_vehicle. This parameter is set to e.g. quad or jsbsim, which seems strange for two such fundamentally different properties. Would it not make more sense to have one parameter controlling what simulator to run (JSBSim, last_letter...) and then another for what model/configuration to run (quad, hexa, Rascal...)?

magicrub, 2015-10-26:
There are several simulators supported, for specific aircraft I would not
suggest jsbsim as we have applied it. There is also ROS, CCRCSim, and
last_letter.

http://dev.ardupilot.com/wiki/sitl-simulator-software-in-the-loop/

krisgry, 2015-10-26:
Thanks for the quick feedback. last_letter seems like a very promising simulator, but it is not quite there yet. E.g. it seems to be missing the possibility to have nonlinear aerodynamic coefficients (other than lift and drag). I will have a closer look at converting our JSBSim model into a last_letter model.

But what you are really saying is that JSBSim will be depreciated sometime in the (near) future?

Georacer, 2015-10-26:
Hello krisgry,

I am the author of last_letter. I think I'd drop by and tell you the
current state of development, since, as you noticed, last_letter is not
there yet when it comes to ArduPlane simulation.

The main problem is that, in its current iteration, the link with SITL is
stuttering and offering very low update rates. As a standalone simulator it
is much more usable.
On the other hand, it's so modular that, if you have a specific
aerodynamics implementation in mind, you could implement it relatively
easily. Check the aerodynamicsLib class:
https://github.com/Georacer/last_letter/blob/master/last_letter/include/aerodynamicsLib.hpp

However, currently I'm juggling two jobs and have let last_letter
development slip (Sorry TeamTilrotor and Mark fhedberg!). As a result, I
don't know when I will be able to code seriously for the project.

As a side note, I'm seriously thinking about using the v-rep engine for the
basis of last_letter, alongside ROS.

I hope this helps you. Feel free to ask further.

All the best,
George

magicrub, 2015-10-26:
There is no plan to remove jsbsim support, it is currently the workhorse
for testing

krisgry, 2015-10-26:
Great to hear your input Georacer. I was really pleased when I noticed that you have used the same notation as Beard & McClain. As it is slightly off the original topic of this issue, I will contact you elsewhere about some questions concerning the future development of last_letter.

Is the problem with the stuttering SITL link last_letter specific, or is it also a problem with JSBSim? As we have no problems with JSBSim, I don't see the point in making the transition yet. (But I see some of the advantages it can bring in the future)

@magicrub So would it be a bad idea if I implement the aforementioned changes and make a pull request? It would not involve a lot of changes in the code, and would be very beneficial for our project since we can stay in sync with the ardupilot master branch at all times. We do all our autopilot/guidance work on a different platform anyways, so no need to change the AP code for that. 

And sorry for being a noob. I'm not used to open source dev at such a large scale :)

magicrub, 2015-10-26:
All Pull-Requests welcome! Tridge is the one to talk to about integrating
into the current sim. He's the gatekeeper.

Georacer, 2015-10-26:
krsigry,

I'm not sure what the problem is and it's been some months since I last
checked on it. After talking with Tridge we had setup last_letter to only
perform time steps after having received control inputs from the SITL. This
was to make sure that the SITL can keep up with the simulation as well as
control time evolution (eg pause etc). However, for some reason (not sure
on which end), one of the two softwares was sending its messages in a
slower rate, which was slowing things to a crawl.
My end of the deal is here:
https://github.com/Georacer/last_letter/blob/master/last_letter/scripts/fdmUDPSend.py

However, you there is the possibility to run a free-running simulation with
SITL, with little tinkering of the code (basically launching parameters).

JSBSim is working without issues, AFAIK and (I think) has native console
support for manual time-stepping. However, I think that it has a different
type of backend in the SITL code.

George

tridge, 2015-10-26:
hi Krisgry,
I've added support for different JSBSim models to master, and updated the docs here:
http://dev.ardupilot.com/wiki/setting-up-sitl-on-linux/#using_a_different_jsbsim_model
You just use the -f option, like this:
sim_vehicle.sh -f jsbsim:MyModel
that matches the syntax I added for last_letter a while ago

krisgry, 2015-10-27:
Wow! And I don't even have to lift a finger? Thanks!

The only thing I am missing from my aforementioned suggestion is adding the following lines to the gitignore file to allow users to add models to the aircraft folders without affecting the ardupilot repo:
/Tools/autotest/aircraft
!/Tools/autotest/aircraft/Rascal
!/Tools/autotest/aircraft/arducopter

When/if new models are added to the official ardupilot repo, they can be removed ignored from the gitignore.

However not having this is not a big problem, as the potential new models can stay untracked.

WickedShell, 2017-02-10:
Closing for now, as I think actually warning that models are untracked is actually a good behavior, as if it is a good model for testing it helps to make it available, and remind you that you might be losing that model in your fork if you don't commit it.

closed 2017-02-10 by WickedShell

ks9354s, 2017-11-08:
hi