ArduPilot/ardupilot #3084: PX4Flow only works after boot via USB connection

dutchrazor, 2015-10-26:
Hey all,

As discussed in this forum thread, some PX4Flow sensors only work after the Pixhawk has first been powered via USB, and only then via battery once the sensor is up and running. ( http://ardupilot.com/forum/viewtopic.php?f=56&t=11832&p=32125&hilit=flow+px4flow+usb#p32125)

Apparently this is fixed in PX4 ( I think by rebooting the PX4flow driver after a some delay), could this also be implemented in Arducopter?
https://github.com/PX4/Firmware/issues/2145

Best regards,

Fabien

Horsj, 2015-11-11:
I am seeing this problem too.  I have managed to get the PX4Flow configured on my hexa, but it is not practical to use it out in the field if I have to power it through the USB before plugging the battery in.

Something I have noticed, which may help with debugging this problem: If I enable the PX4Flow and power power up with the main battery first, then connect the Mission Planner, the altimeter on the Flight Data page is climbing very rapidly (it goes up thousands of feet and doesn't stop), although the hexa is sat on my bench. After doing this, I checked the altitude in the logs and it appears normal ie. doesn't reflect what I am seeing on the Flight Data page.

moon-siama, 2015-12-14:
It is a same fix of PX4/Firmware#2145

ardupilot/mk/PX4/ROMFS/init.d/rc.APM --> APM startup script for NuttX on PX4

and add sleep 4 before the PX4Flow start

#optional PX4Flow sensor

##### sleep 4
if [ -f /bin/px4flow ]
then
  if px4flow start
  then
    echo "Found px4flow sensor"
  fi
fi

And you can debug it with Nsh

Horsj, 2015-12-18:
I created the file /fs/microsd/etc/extras.txt with the following content:

if [ -f /bin/px4flow ]
then
sleep 4
if px4flow start
then
echo "Found px4flow sensor"
fi
fi

If I do a normal power up, px4flow is still not starting, but if I go in with nsh and run my script manually: sh /fs/microsd/etc/extras.txt then px4flow starts with: [PX4FLOW] on I2C bus 1 at 0x42 (bus: 100KHz, max: 400KHz)and I get the message from the end of the script: Found px4flow sensor.  I'm quite familiar with UNIX type operating systems, but am just finding my way around NuttX. It looks to me like extras.txt might not be being executed; Is this likely and what might be some reasons for it not executing? - I could be barking up the wrong tree, but I'm at a bit of a loss.

Naterater, 2019-01-07:
I'll assume that the lack of responses here since @rmackay9's merged PR means that this was successful! 🥇  If not, please re-open this issue.

closed 2019-01-07 by Naterater