ArduPilot/ardupilot #3088: Thesis student looking for security analyst

KarelD, 2015-10-27:
Hi,
I am a computer science student (KU Leuven - Belgium), specialization secure software.
For my thesis, I will perform a security analysis of the software/firmware running on drones. After a lot of reading, I have found that there has been a lot of research on the hardware and communication aspect (Wifi-hacks, GPS spoofing, bluetooth, Zigbee, ...), but it seems like much of the secure software part has been untouched. I have found some articles about DARPA and Galois creating hack-proof drone software (commercial). I am willing to further investigate whether or not some common vulnerabilities are present, whether or not they can be exploited and what the consequences can be, particular on open source software (ref. Dronecode project). Is there someone who can bring me into contact with a researcher or interested person who has done some research in the field of security of drone software?

I would like to further refine the topic or goal depending on the 'demands' of the community and be up to date with the current state of the art on the safety of drone software.

Best regards,
Karel

squilter, 2015-10-27:
Hi Karel,

1. Mavlink does not have encryption or authentication built into the protocol.  So it can be very easy to inject packets and take over control.  It really depends on the transport layer.  If Mavlink is running on a secure wifi network, then it will be very difficult.  If it is running on unencrypted 915 mhz radios, then it is easy.

2.  GPS spoofing is always easy.  Hopefully the vehicle would notice that something is wrong, and it would just failsafe and land.  I don't think it would be possible to spoof GPS to redirect the vehicle to go somewhere else.  Hard to test, because it's very illegal to test.

This isn't really the place to bring this up, so maybe post to diydrones?  Not sure if this is appropriate for drones-discuss.  I'd be happy to respond wherever you post this to, though.

closed 2015-10-27 by squilter

KarelD, 2015-10-27:
Hi Squilter,

First of all many thanks for the response. I did post my question on DIYDrones first (in "my project"), but in anticipation of a response, I posted it here too. Maybe this is indeed not the right place to bring it up, but my thesis is going to be completely about drones (their firmware).

As you say that the security of Mavlink depends on the transport layer, it could be an idea to de research about the inclusion of encryption in Mavlink. No idea whether or not this has been investigated before, but I will take a look at it.

About GPS spoofing, there are many research papers discussing this topic and as you say it's illegal, so this will be out of scope of the thesis.

Once again thank you for the response and I hope to further talk about it on another forum.

squilter, 2015-10-27:
Yes, authentication is being considered for the next version of mavlink on the [tsc mailing list](https://lists.dronecode.org/mailman/listinfo/dronecode-tsc).  Not encryption.  Can you post the link to the diydrones post, so we can keep talking about it there?

rmackay9, 2015-10-28:
Yes, coincidentally, Tridge was talking about MAVLink and encription just yesterday.

KarelD, 2015-10-29:
[link to DIY] (http://diydrones.com/forum/topics/computer-science-thesis-student-looking-for-security-analyst?xg_source=activity)

There you go. I've found that they issued a RFC about Secure MAVLink in 2013. In Another project, Galois added encryption to Mavlink to protect against forgery, replay attacks, and snooping. Did not yet find the time to read it in detail, but will do it soon. [Link to paper] (http://smaccmpilot.org/artifacts/Galois-commsec.pdf).