ArduPilot/ardupilot #3098: Accelerometer calibration fails in 3.4.0 stable

jankansky, 2015-10-29:
I can't get accelerometer calibration to work in ArduPlane 3.4.0 stable.

Here is the pixhawk config:

nsh> \0x1b[Kver all
HW arch: PX4FMU_V2
FW git-hash: 3aa9eb0c2eeb489874f4532653ede5ec9ea3f9b6
Build datetime: Sep 24 2015 00:43:46
Toolchain: 4.7.4 20140401 (release) [ARM/embedded-4_7-branch revision 209195]
MCU: STM32F42x, rev. 1

WARNING   WARNING   WARNING!
Revision 1 has a silicon errata
This device can only utilize a maximum of 1MB flash safely!
http://px4.io/help/errata

UID: 520036:30345102:37383933 

I have reset all EEPROM values to their default values with the exception of the SYSID_THISMAV, which I set to 4, and BRD_SAFETYENABLE which I set to zero since we want high rate log recordings to start on power on, and are just using the pixhawk as a logger in this application.

If I issue a 3D Accelerometer calibration via apmplanner2 on Linux, I get a message that says CALIBRATE MAV004, and then it times out after 40 seconds.

If I issue a accelcal via mavproxy.py version 1.4.34, I get:
:~> mavproxy.py --master=/dev/ttyACM1
Connect /dev/ttyACM1 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyACM1
 MAV> ~=online system 4
RTL> Mode RTL

APM: ArduPlane V3.4.0 (3aa9eb0c)
APM: PX4: c4f39028 NuttX: d48fa307
APM: PX4v2 00520036 30345102 37383933
Received 584 parameters
Saved 584 parameters to mav.parm

RTL> 
RTL> accelcal
RTL> APM: command received: 
Init GyroAPM: Initialising APM...
**
APM: Place vehicle level and press any key.
 APM: Initialising APM...
APM: Initialising APM...
APM: Initialising APM...
APM: Initialising APM...
APM: Initialising APM...
APM: Initialising APM...
Timed out waiting for user response
APM: Calibration FAILED
Got MAVLink msg: COMMAND_ACK {command : 241, result : 4}

The pixhawk nuttx shell shows the following errors during calibration:
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17
Overtime in task 17

Thanks for any help!
-Jan

jankansky, 2015-10-30:
I was able to calibrate using mavproxy.py.  The problem is the following:

The following statement is inaccurate.
APM: Place vehicle level and press any key.

You must hit <enter>.  Simply hitting the space bar, or any other key is not enough to trigger the next stage of the calibration.

As to why I can't calibrate using newer revisions of apmplanner2, that remains unsolved.

yazinsai, 2016-05-30:
@jankansky how did you end up solving this? would really appreciate some help, as I have the exact same issue. thanks!

WickedShell, 2017-11-03:
This is a support issue rather then an autopilot issue, and there is a better (less error prone) protocol for controlling accelerometer calibration now as well.

closed 2017-11-03 by WickedShell