ArduPilot/ardupilot #3107: DJI CANBUS Support

artemen, 2015-10-31:
DJI CANBUS Support will be a great addition to use with DJI gimbals.

R-Lefebvre, 2015-10-31:
Do we have any info on the protocol?  Or is it logic analyzer time?

artemen, 2015-10-31:
Should be pretty easy as there are already clones of iosd, pmu and GPS all of which are canhub based.

helvaci, 2015-11-30:
There is some people on this page makes a progress:
http://www.rcgroups.com/forums/showthread.php?t=2071772
It will be cool but i think it will be add visibly load to processor.

EShamaev, 2017-06-27:
Is this still actual?
If the hardware can be provided, I can take over to implement the feature.

rmackay9, 2017-06-27:
I think ardupilot would provide funding for the hardware if someone can point out which hardware would be good.

proficnc, 2017-06-27:
I would suggest an M600
It would be a great demo platform, with a cube!

auturgy, 2018-01-29:
Cross linking to dapo
https://discuss.ardupilot.org/t/code-bounty-for-dji-canbus/22813/9

geofrancis, 2018-02-04:
You should check out cosmostreamer it's basically using a pi to interface with an inspire1 camera/gimbal

jasin755, 2019-11-05:
+1 Controlling Ronin MX is best usage of CAN bus on Ardupilot

alxbouchard, 2020-11-21:
There's a new SDK for the Ronin R series:
https://terra-1-g.djicdn.com/851d20f7b9f64838a34cd02351370894/DJI%20R%20SDK/DJI_R_SDK.zip

Wodie, 2021-04-15:
I been debugging data frames from Ronin S, is anyone working with it? At the moment I am only missing the 16 bit CRC to have it working.

ayi-afk, 2021-05-20:
I tried to do all like in doc :( but nothing happens, even sending the example packet from SDK doc doesn't work ... any1 got it working over CAN?

ceinem, 2021-05-27:
@ayi-afk same for me, didn't get any movements or reactions at all from the gimbal over the new SDK. Did anyone have any luck?

reallycoyote, 2022-04-27:
have you looked at the sdk documentation for commands and are you using it in conjunction with the focus wheel? i read that no data is sent unless an accessory is attached. https://usermanual.wiki/m/35b7e4af8653ff4d7d14e5d9ec3ced9e3e085a0d1ffcab964db6bc053ecd0aaf.pdf

ceinem, 2022-04-27:
I got it working at some point for the RS2, wrote a minimal ROS integration which you can find [here](https://github.com/ceinem/dji_rs2_ros_controller)

reallycoyote, 2022-04-27:
cool great. i'll look into it. just out of curiosity, were you able to read from the gimbal as well such as position speed etc. or only send? I'm working on something to log and replay for motion control.

ceinem, 2022-04-27:
I was also able to read the data, at least position. I'm not sure about speed, I would have to look into the code again. 
The code is heavily based on ROS and ROS-CAN integrations, but during development I also had it working with python-can and without any ROS integrations, so should be easy to adapt. 
Error correction  and those things are heavily based on the sample SDK code of DJI.

reallycoyote, 2022-04-27:
thanks! this is giving me lots of hope for py-can/can-utils

arcusfelis, 2022-05-16:
Hi, nice project.
@ceinem is it possible to control follow focus motor using CAN?

ceinem, 2022-05-16:
with the code I have implemented so far: no. In general: maybe? you will have to look through the rSDK documentation and see if there are some specified commands for that. 
If yes, then it is easy to implement.

arcusfelis, 2022-05-16:
@ceinem yea, there are documented commands (I am surprised, actually).

https://usermanual.wiki/m/35b7e4af8653ff4d7d14e5d9ec3ced9e3e085a0d1ffcab964db6bc053ecd0aaf.pdf
Section 2.3.4.19 Focus Motor Control Command

It is "rotate to absolute position". Sad, there is no rotate by one step of the stepper motor (And I am not surprised, actually, because of how poor the api is). I am wondering how absolute positions work without calibration now (time to get an arduino and do some coding I guess).

ceinem, 2022-05-16:
Oh, interesting. I still had an older version of the document without that command. Yeah, the api is quite poor. Maybe the focus motor position is reported with all the position and speed status messages? 
I don't have a focus motor, so I can't test that unfortunately.

arcusfelis, 2022-05-16:
yea, I have to sniff what the wheel is sending when I move it (something says it sends relative move, not absolute).

ceinem, 2022-05-16:
hm, do you mean the communication between wheel and gimbal? Maybe that's s-bus and not CAN, then it would probably only be relative.

reallycoyote, 2022-05-16:
prototyping this now

reallycoyote, 2022-09-24:
I'm about to do more heavy prototyping on this but I am waiting on some
pieces. I was wondering did anyone log or sniff the can bus without sending
commands and doing some movements on the gimbal? If not can you? What is
the result like?

arcusfelis, 2022-09-24:
Yea, managed to make canbus working with Arduino and a canbus shield, some
c coding is required to assemble all packages and build them.

The API is very limited though.
You can see communication of the wheel on the bus, but that is about it.
And the communication of the wheel is not documented.

There are also some packages flying around on the bus, undocumented.
Everything else you cannot trace because it is send on a separate bus
internally.

I think sniffing will not help much because packages I've seen use custom
checksums which is not easy to reverse engineer.

So, yea, only canbus commands from that SDK API are available to work with.

reallycoyote, 2022-09-25:
I have this theory I want to test but as I said I'm waiting for my parts.
Is it possible to run python-can logging and then use force mobile on a
phone to move the gimbal around panning and tilting. Then stop the logging
and then replay the log over the canbus to hopefully replay the same
movement?

arcusfelis, 2022-09-25:
you can query for gimbal position using SDK API.

```
- Handheld Gimbal Information Obtaining
Reply frame
2 2 yaw int16_t yaw axis angle (unit: 0.1°)
4 2 roll int16_t roll axis angle (unit: 0.1°)
6 2 pitch int16_t pitch axis angle (unit: 0.1°)
```

and after you can set the values back.

You will not see anything on a bus, because force mobile and gimbal operations are communicated on the internal bus, not the one you can connect to. So yea, replay would not work.

reallycoyote, 2022-09-25:
Gotcha. I am looking at Bluetooth log replay now as well

rmackay9, 2022-10-31:
Some updates on this

- [AP funding proposal to purchase an RS2 for driver development](https://discuss.ardupilot.org/t/approved-used-dji-rs2-for-driver-development/91829)
- [Discuss topic re adding support for the RS2 gimbal](https://discuss.ardupilot.org/t/ap-support-for-dji-rs2-rsc2-rs3-gimbals-maybe/91830)

crazyfrog-github, 2022-11-18:
> Gotcha. I am looking at Bluetooth log replay now as well
> […](#)

Hi,

I am trying to control the RS2 Gimbal exactly like the DJI Assist app with Force Mobile via Bluetooth.

https://www.youtube.com/watch?v=0CQ4LSb0Evw

The RS2 Gimbal has one Bluetooth service 0000fff5 which receives and executes the command, I want to send continual write commands to control the movement of gimbal. I managed to find the UUID that performs a recentre and what makes it go to the right, left, up and down using a Bluetooth sniffer and Wireshark in between the DJI Assist app on an Android phone and RS2 to identify the service and the UUID or hex being sent to the RS2.

Device Service:

WRITEABLE WITHOUT RESPONSE service UUID (0000fff5-0000-1000-8000-00805f9b34fb)

I was wondering if you would help me discover if there is a correlation between the DJI R SDK Protocol and User Interface hex commands and if they are sent by the DJI Assist App via Bluetooth service above to the RS2 to control it.

You have experience via the CANBus side of things and thought you could help shed some light here.

reallycoyote, 2022-11-26:
Hello friends Cornelius and I and another have written a simple python
control script - movement and focus we will release it at Christmas 🎄⛄🎅
on git

I posted on Reddit and dji forums so you can follow the updates. I'm
resting after putting it through the battlefield and cleaning up the code.
Talk to you soon
xR

https://www.reddit.com/r/dji/comments/z54qjd/myself_and_collaborators_have_successfully/

crazyfrog-github, 2022-11-26:
Hey,

Looks good, I assume your code can move the DJI RS2 gimbal faster with pan motion than in the video which is more like Tai Chi than King Fu?

Thanks

Paul

https://www.reddit.com/r/dji/comments/z54qjd/myself_and_collaborators_have_successfully/

reallycoyote, 2022-11-26:
find out for yourself in a month

crazyfrog-github, 2022-11-26:
Okay thanks, looking forward to the big reveal.

reallycoyote, 2022-12-06:
merry christmas

https://github.com/rileycoyote87/DJI-Ronin-RS2-Log-and-Replay/
https://www.reddit.com/r/dji/com ... _have_successfully/
<https://www.reddit.com/r/dji/comments/z54qjd/myself_and_collaborators_have_successfully/>

ho ho ho

rmackay9, 2023-02-20:
Support for the DJI RS2/RS3 Pro gimbals are in "latest" and will be released with 4.4.0 which should start beta testing within a couple of weeks.

[We've also got a wiki page here which describes the setup and configuration](https://ardupilot.org/copter/docs/common-djirs2-gimbal.html)

electricpocket, 2023-04-05:
> > Gotcha. I am looking at Bluetooth log replay now as well
> 
> Hi,
> 
> I am trying to control the RS2 Gimbal exactly like the DJI Assist app with Force Mobile via Bluetooth.
> 
> https://www.youtube.com/watch?v=0CQ4LSb0Evw
> 
> The RS2 Gimbal has one Bluetooth service 0000fff5 which receives and executes the command, I want to send continual write commands to control the movement of gimbal. I managed to find the UUID that performs a recentre and what makes it go to the right, left, up and down and the UUID or hex being sent to the RS2.
> 
> Device Service:
> 
> WRITEABLE WITHOUT RESPONSE service UUID (0000fff5-0000-1000-8000-00805f9b34fb)
> 
> I was wondering if you would help me discover if there is a correlation between the DJI R SDK Protocol and User Interface hex commands and if they are sent by the DJI Assist App via Bluetooth service above to the RS2 to control it.
> 
> You have experience via the CANBus side of things and thought you could help shed some light here.

Hi, have you made any more progress on discovering the Bluetooth services and properties to control the RS2? I have an RS3 Mini and would be interested in helping getting Bluetooth remote control working so I can program it

peterbarker, 2024-07-31:
Well, we have support in ArduPilot for this, so closing.

@inspired27 - @electricpocket asked you a question which you didn't respond to, this is a last ditch on that front :-)

closed 2024-07-31 by peterbarker