ArduPilot/ardupilot #3108: Mavlink Stopped Working Over 3dr Radio

davidbitton, 2015-10-31:
MP no longer detects a heartbeat when connecting over a 3dr radio from Win7x86 to a Pixhawk. The radios have solid green lights indicating an established link between them. I am able to communicate to both radios from the 3dr radio settings window in MP. SERIAL1_BAUD is set to 57 and SERIAL1_PROTOCOL is set to 1. I am able to open COM3 at 57600, 8/n/1, however, the "text" showing up in the window is garbled. What is a Mavlink heartbeat packet supposed to look like? Thanks!

magicrub, 2015-10-31:
This doesn't sound like an ArduPilot issue, please post to https://github.com/diydrones/MissionPlanner/issues instead. Would be helpful if you included information such as what versions you're running.

closed 2015-10-31 by magicrub

davidbitton, 2015-10-31:
As a follow-up, it was a config issue; BRD_SER2_RTSCTS was changed in error. Thanks.