ArduPilot/ardupilot #3110: Disarm_Delay parameter confusing if used with Motor Interlock function.

R-Lefebvre, 2015-11-02:
When Motor Interlock function was created for multirotors, the choice was made to reduce the hard-coded disarm delay from 10 seconds to 5 seconds when Motor Interlock function is used, since the ground state is more dangerous when armed, but Interlock switch is "low".

Then when Tridge elected to create the DISARM_DELAY parameter, he attempted to keep the logic equivalent, defaulting this to 10 seconds, but then cutting it in half if Motor Interlock is enabled.

This now makes Disarm_Delay parameter confusing, since the time is cut in half if using Interlock.

Recommend removing the div/2 of the time, and suggesting to users using Motor Interlock to simply use a shorter Disarm_Delay time.

magicrub, 2015-11-02:
Not to be confused with the similarly named but totally different Plane
param LAND_DISARMDELAY
<http://plane.ardupilot.com/wiki/arduplane-parameters/#landing_disarm_delay_arduplaneland_disarmdelay>

R-Lefebvre, 2015-11-02:
Does Disarm Delay have any function on Planes?

magicrub, 2015-11-02:
There is no equivalent in plane (and vice versa)

R-Lefebvre, 2015-12-21:
This issue was resolved with this commit:

https://github.com/diydrones/ardupilot/commit/c968ec3a63767f714247422a990066807dd80dd1

closed 2015-12-21 by R-Lefebvre