ArduPilot/ardupilot #3126: Copter: first gyro call can cause GPS driver to start auto-detect?

rmackay9, 2015-11-05:
This log file is of an event in which GPS is lost almost immediately after arming (and arming happened only 21 seconds after plugging in the battery).
https://dl.dropboxusercontent.com/u/3852647/Copter-3.3-GPSlossStartup/12.BIN

There are no GPS messages in the log for 7 seconds which I suspect means the GPS auto-detect started running.  I suspect that the auto-detect which normally occurs if no messages can be understood for a few seconds can also be triggered by the delay during the first arming's gyro calibration.

WickedShell, 2016-03-15:
The autodetect window has been widened to two seconds, if we need longer time periods we will need to create a way for the vehicles to tell the GPS driver to not panic and unload a driver after coming back from a calibration.

WickedShell, 2017-01-13:
Closing as this seems to be solved, or at least no one has reported this as still being a current issue.

closed 2017-01-13 by WickedShell