ArduPilot/ardupilot #3133: Please add capacity consumed (mAh) to mavlink

AndKe, 2015-11-06:
If we could see capacity spent, (mAh) - then we could easily calculate remaining power into remaining flight time in GCS, and know if a mission exceeds battery capacity well ahead. (when for example flying pattern in strong crosswind)

Calculating mAh from current and voltage in GCS is not precise at low data rates.

Mavlink BATTERY_STATUS have the option for capacity used, or maybe another message can be expanded with that , like SYS_STATUS

rmackay9, 2015-11-07:
@AndKe, we can't easily expand existing messages but it looks like the BATTERY_STATUS message you've mention has what you need (or something that will allow the GCSs to calculate what you're requesting).  So if we added support for BATTERY_STATUS this would be good enough?
http://mavlink.org/messages/common#BATTERY_STATUS

AndKe, 2015-11-07:
Yes, if nothing else, "current_consumed" of BATTERY_STATUS would alone be sufficient to properly estimate energy spent, and remaining flight time based on battery_capacity parameter, and the completed percentage of a mission.

magicrub, 2015-11-07:
Is that implemented? If so, let's close this

AndKe, 2015-11-07:
not implemented, that's what I ask for in the subject.

rmackay9, 2015-11-08:
Great, thanks for the clarification.  updated subject line a bit.

lvale, 2015-11-10:
There is already a % capacity used of battery on MSG #1 sent as a percentage http://mavlink.org/messages/common#SYS_STATUS, besides what can be sent on MSG#147

What is missing is to know what is the original battery capacity even from MSG #147, which can only be retrieved from the parameters, which the user SHOULD configure specifically, but sometimes don't....

AndKe, 2015-11-10:
that's right, I may have used "battery_remaining" (byte) from SYS_STATUS , but would get higher precision on the estimate if we had mAh 
-which would provide much higher resolution and be able to update smoothly, not only when a percent does, (once in ~90seconds)
An estimated flight time will be jumping a lot when calculated from often-updated distance flown, and a percentage that updates less then once a minute.

AndKe, 2015-11-10:
for the same reason - I miss a "total_travelled_distance" report from UAV...  how can we proceed to add it ?  I think we all agree that "capacity spent so far", "total available capacity" , "travelled distance" and "total mission length" , "distance to home"  are significant data that would allow to estimate precisely the UAV capability , and contribute to improved safety.

lvale, 2015-11-10:
@AndKe Sorry, but we do already have "total travelled distance". It's a simple calculation obtained from every GPS position on the logs, or even on the telemetry stream. The GCS can do that calculation easily.
It's a question if the telemetry data stream is only supplying relevant data (as raw as possible) that can be processed at a ground station (where there is typically more cpu power) or spend cpu cycles on board.

btw: the battery percentage does vary at a higher rate than 90s. It depends on consumed power...

AndKe, 2015-11-10:
@lvale  Yes, I do agree that we need to keep the telemetry link lean and effective.

But the autopilot is calculating mAh consumed, and reporting a percentage *because* if we just delivered lots of voltage and current measurements, we could do the same at GCS - but at a much less precision.  
The same way; if we do total distance calculation on board - then it can be reported to GCS as 0.5Hz , and still be very precise.
The final problem with letting GCS do the distance calculation, is that any telemetry drop could influence the quality, and the solution is not useful at all for iridium or other low-bandwidth links.

But it we do the suggested calculations onboard (we have/will have plenty of CPU power anyway) - we can send them at longer intervals, and still have a correct and precise energy estimate.

R-Lefebvre, 2015-11-10:
Luis, how do we see the total distance travelled?  Is that in flash log only, or can it be seen live in the GCS?

lvale, 2015-11-10:
Hi @R-Lefebvre 
If you have 2 consecutive GPS positions on the log we just have to determine the distance between them and sum add to a totalizer. It could be done on a GCS also.

R-Lefebvre, 2015-11-10:
Ok, I thought you were saying this is already done/available.

lvale, 2015-11-10:
I wrote that "it's a simple calculation" :)
http://www.movable-type.co.uk/scripts/latlong-vincenty.html

AndKe, 2015-11-10:
auotpilot is already doing a distance calculation "all the time" while flying with "camera trigger by distance"  - so it's only a matter of adding the distance to a uint32 total_distance  - and make the camera function read and compare that, instead of doing it all over again - CPU load will not be greater than flying with trigging by distance.   
all we need is to keep those before mentioned  variables on board, and send them in an MAVlink packet now and then - added safety will then work regardless of telemetry link speed/quality.

amilcarlucas, 2017-09-25:
We do have a BATTERY_STATUS message now.

WickedShell, 2017-09-26:
Given that the issue is about BATTERY_STATUS (which @amilcarlucas is correct is supported and sends the requested fields) and the rest of the conversation seems to be application related (and off topic) so closing. If there was other content that isn't addressed and (not already another issue) feel free to open a new issue.

closed 2017-09-26 by WickedShell