ArduPilot/ardupilot #3135: ArduPlane 3.4.0 arming issue while start for APM2

mariansoban, 2015-11-07:
If parameter ARMING_REQUIRE is set to value '1', throttle channel value is directly passed through while ArduPlane 3.4.0 at APM is starting. I find this bug quite dangerous.
It looks that it's solved for Pixhawk on lines 82-84 in radio.cpp (https://github.com/diydrones/ardupilot/blob/ArduPlane-release-AVR/ArduPlane/radio.cpp).
Probable workaround for APM board is to set parameter ARMING_REQUIRE to value '2' - ZERO PWM for throttle while disarmed (lines 68-70 in same radio.cpp).

gmorph, 2015-12-18:
Note this only affects Plane users running APM2. 
Thanks, Grant.

gmorph, 2016-09-06:
This is actually fixed in the AVR branch but I did it after 3.4.0 was released.  So we would need to release another version of APM Plane for this fixed to be picked up and as I recall the build is not straighforward as there is some trickery required to make it "fit" memory wise.

Thanks, Grant.

WickedShell, 2016-09-06:
@gmorph if that this is addressed code wise already and then can this issue be closed?

gmorph, 2016-09-06:
Yes fine to close from my perspective but at this stage we have no plane to do a formal release with that fix in it.

magicrub, 2016-09-06:
Grant, can you post the commit that fixes this?

closed 2016-09-06 by magicrub

mariansoban, 2016-09-07:
@gmorph is the fix the one from thread: http://discuss.ardupilot.org/t/warning-danger-motor-starts-even-when-disarmed-before-boot/7029/5 ? (change of "true" -> "false" on line https://github.com/ArduPilot/ardupilot/blob/ArduPlane-release-AVR/libraries/AP_HAL/Util.h#L12)
If yes, then there is another minor issue @ APM2 with PWM pulse width on channel 3 during boot - width is  just 0.4 ms which my ESCs don't like. They are beeping all the time while APM2 is booting .