ArduPilot/ardupilot #3149: Plane:Does ArduPlane:ALT_MIX still used for Plane?

jinchengde, 2015-11-10:
I check the code, there is no "altitude_mix" in update_alt function. so does this parameter still work now? recently, we need to used RTK GPS for Plane and used GPS altitude to replace the Baro Alt.

I'm not clear how NavEKF handle this issue. does anyone have suggestion?

magicrub, 2015-11-10:
Due to GPS alt being very inaccurate in Australia without sbas, it was
decided long ago that baro should always be used instead of gps so that
feature had been depreciated. Now with RTK, GPS is more accurate than baro
especially over time so it needs to go back in.

rmackay9, 2015-11-10:
i think it would all end up being done inside the EKF though.  The EKF does use the GPS altitude (and velocity i think) and I wouldn't be surprised if there are params that control the "mixing".

WickedShell, 2016-04-18:
ALT_MIX is a dead parameter, to use GPS altitude set EK2_ALT_SOURCE to 2 (see http://ardupilot.org/plane/docs/parameters.html#ek2-alt-source-primary-height-source )

closed 2016-04-18 by WickedShell