ArduPilot/ardupilot #3160: add branching model link to wiki page

sarnold, 2015-11-11:
Having this link:  https://pixhawk.org/dev/branching_model

on the wiki would be very helpful.  Thanks.

magicrub, 2015-11-11:
That is PX4 specific, not ArduPilot.

SJLC, 2015-11-11:
Is there anywhere the branch strategy for ArduPilot repo *is* documented?

sarnold, 2015-11-11:
Appears to follow the same model (or very close/similar one).  What are the specific differences if any?

rmackay9, 2015-11-12:
We have this wiki page on Copter releases procedures. http://dev.ardupilot.com/wiki/release-procedures/

We have master which is the latest development version (i.e. bleeding edge) and may or may not have been flown by anyone.
The official release (and all past releases) are sitting in an appropriately named branch (i.e. "ArduCopter-3.2.1" or "Copter-3.3").
beta releases are controlled via a tag (i.e. ArduCopter-beta) and may be done off master or off the branch prepared for the next version.

sarnold, 2015-11-12:
Okay, thanks for the info, but it looks like something changed in the workflow between ArduCopter-3.2.x and previous and the Copter-3.3 branch, since I only see tags after that (eg, recent github release tags using version/stable/beta and no 3-digit branches).  I don't mean to nit-pick (I swear!) but we are writing a new ICD and updating a draft interface requirements spec so we need to be specific about versions, messages, etc.  Thanks in advance.

closed 2019-01-07 by Naterater