ArduPilot/ardupilot #3167: SD Card reliability/performance

lvale, 2015-11-12:
There are some reports of users having problems with missing data on logs, specially when using "alternative" PixHawks, mostly because of the "quality" of the SD Cards supplied. I believe that there was also an issue some time ago with the cards from 3DR.

We already have a small test routine at https://github.com/diydrones/ardupilot/blame/master/libraries/DataFlash/examples/DataFlash_test/DataFlash_test.cpp

What I propose is to develop a bit more that routine, or building a new one, and make it accessible via CLI (for now) that would run a series of tests reading/writing data to a card and make the results available to the user, so that he/she can make a decision on the card used on the Flight Controller.

Just had a chat with Peter Barker on this issue.

peterbarker, 2015-11-12:
Tests should include:
 - writing at different rates
 - reading and writing simultaneously (how much data does terrain really read, 'though?)
 - ensuring we read back what we wrote

Results could be put into a file on the SD card.  How would a user then retrieve that without having to pop their SD card?  Could open a special dataflash log file for the results.

We should try to get a Wiki page together showing expected results using specific SD cards and a genuine PixHawk.

peterbarker, 2015-11-14:
WIP branch: diydrones/ardupilot/compare/master...peterbarker:sd-card-test-wip?expand=1

ptsneves, 2015-11-19:
This can be done checksumming writes, either on the Ardupilot itself or on the routine you talked about.

lvale, 2017-01-02:
@peterbarker do we close this ?

peterbarker, 2017-01-02:
On Mon, 2 Jan 2017, Luis Vale Gonçalves wrote:

> @peterbarker do we close this ?

I'm still poking the branch here every now and then.

lvale, 2017-01-03:
Now "officially" assigned :)

closed 2018-08-19 by lvale