ArduPilot/ardupilot #3171: Bad AHRS reported on Pixhawk

chrismcna, 2015-11-13:
I've been diagnosing an issue with ardupilot 3.3.1 on pixhawk, i get "Bad AHRS" and i have diagnosed the problem to being within the class AP_NavEKF.cpp healthy method.

the value EKF3 - IPD (Mission planner log review) or variable innovVelPos[5] spikes above 1 every now and then (highest ive seent is 1.4 while flight controller is sat still on a table which leads to the healthy method returning false. This leads to ardupilot not arming, and an annoying messing saying "Bad AHRS", I believe this is an issue for many users of ardupilot.

I have been posting my diagnoses with a high level of detail at including screen shots of dataflash log graphs
http://ardupilot.com/forum/viewtopic.php?f=111&t=14322&start=15

I have found the issue but I'm unsure of the best way to resolve it? 

The "If" statment in question - Ardupilot 3.3.1 AP_NavEKF.cpp line 456

    // barometer and position innovations must be within limits when on-ground
    float horizErrSq = sq(innovVelPos[3]) + sq(innovVelPos[4]);
    if (!vehicleArmed && (fabsf(innovVelPos[5]) > 1.0f || horizErrSq > 1.0f)) {
        return false;
    }

priseborough, 2015-11-14:
This has been addressed since release of AC3.3.1 with use of a filtered value

https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF/AP_NavEKF_core.cpp#L102

closed 2015-11-14 by priseborough

chrismcna, 2015-11-14:
Cheers,

and keep up the good work :)

SlipstreamFPV, 2016-01-04:
Has this been fixed in Arduplane? I'm still getting Bad AHRS errors way to frequently over nothing.

magicrub, 2016-01-04:
what version are you running?

SlipstreamFPV, 2016-01-05:
3.4 on pixhawk

priseborough, 2016-01-05:
@SlipstreamFPV How do you know it's over nothing? Please provide a data log with pre-arm logging enabled and i will have a look at it.