ArduPilot/ardupilot #3174: Division by zero in AP_NavEKF_core

alexbuyval, 2015-11-14:
After rebase my own branch with master I have faced out with runtime error in the SITL mode in the following line:

https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF/AP_NavEKF_core.cpp#L874

rmackay9, 2015-11-14:
Thanks for the report.  I've ping'd Paul.

priseborough, 2015-11-15:
dtDelVel is a weighted average of the delta time returned by the INS library get_delta_velocity function.

I will put in protection against that, but if the get_delta_velocity function is returning zeros for delta time, then there is another error, either in the INS library, or more likely in the simulator.

priseborough, 2015-11-16:
https://github.com/diydrones/ardupilot/pull/3183

tridge, 2015-11-16:
hi Alex, can you tell us how to reproduce this? It shouldn't be possible

alexbuyval, 2015-11-17:
I have got this error in [ROS/Gazebo SITL mode](http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/). I think that maybe the simulation environment produce invalid data during initialization. This error occures on the first call of UpdateStrapdownEquationsNED method always.

priseborough, 2015-11-29:
This cannot happen with the flight code

closed 2015-11-29 by priseborough