ArduPilot/ardupilot #3175: Runtime error in AP_NavEKF2_core

alexbuyval, 2015-11-14:
I have faced out with the runtime error if I disabled a compass in the following line:

https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L206

I think, we should check the  _ahrs->get_compass() on 'null'

Please note, I am using the ROS SITL mode. I did not check it on real drone.

priseborough, 2015-11-28:
Please update the line reference for current master

alexbuyval, 2015-11-28:
[magSelectIndex = _ahrs->get_compass()->get_primary();](https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L250)

alexbuyval, 2015-11-28:
Also I have got the same runtime error in AP_NavEKF2_Measurements.cpp in the following line:
[uint8_t maxCount = _ahrs->get_compass()->get_count();] (https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_Measurements.cpp#L164)

priseborough, 2015-11-29:
This is not an EKF issue. That routine should not be able to return null.

closed 2015-11-29 by priseborough