ArduPilot/ardupilot #3179: Bad AHRS warning too strict?

GrahamDyer, 2015-11-14:
A number of people (including myself) are getting the Bad AHRS warning on our Pixhawk or AUAV-X2 boards. Are the checks for this perhaps a little too strict?
Ref: http://diydrones.com/forum/topics/bad-ahrs

magicrub, 2015-11-14:
Are you, and the others, running the latest release? It's much better now.

R-Lefebvre, 2015-11-15:
I get it way too frequently too.

GrahamDyer, 2015-11-15:
Running AP 3.4 and AC 3.3, getting on both

severinleuenberger, 2015-11-15:
Running APM 3.4 plane and see it to often as well

OXINARF, 2015-11-15:
Maybe related to #3171?

rmackay9, 2015-11-16:
Yes, I suspect @OXINARF is right.  Maybe we should add this to the list for a Copter-3.3.2 patch release.

magicrub, 2015-11-16:
And plane v3.4.1

R-Lefebvre, 2015-11-16:
But if I'm reading it right, according to Paul this was already fixed in 3.3.1?

OXINARF, 2015-11-16:
No, it was fixed after the 3.3.1 release.

R-Lefebvre, 2015-11-16:
Ah, I see.  I read "since the release" the wrong way.

OXINARF, 2015-11-16:
To supply more info: the commit that Paul said that fixed it should be 59df23b, which came from the PR #3101

mtcl, 2015-11-18:
Same issue here, I am getting Bad AHRS error also, this happened so much while flying hat it made the copter crash. Its a brand new Pixhawk's second flight.

rmackay9, 2015-11-18:
@mtcl, this should cause crashes so it would be good to provide a dataflash log.

Zanquar, 2016-01-07:
calibrate Accel again and again, and then, the warning can be solved

magicrub, 2016-01-07:
Not all IMUs are created equal (mfgr error), nor are the calibrations (human error). I've added a param "INS_ACCTHRESH" to help loosen it for mfgr type error. https://github.com/diydrones/ardupilot/pull/3364
Is that good enough to close this or is more work needed to fix the root cause?

Also, EKF2 is more tolerant (it can compensate faster/better) so the thresholds could be loosened when EKF2 is used.

courierdude, 2016-02-04:
Has anybody found a solution to BAD AHRS? Please, I'm begging you. This is my third day of re installing firmware, doing ACCEL calibrations, GPS calibrations, Googling, and generally pulling my hair out.

magicrub, 2016-02-04:
fixed with https://github.com/diydrones/ardupilot/pull/3364 on Plane v3.5.0

closed 2016-02-04 by magicrub