ArduPilot/ardupilot #3187: Download select subset of DF log over mavlink

WickedShell, 2015-11-16:
When downloading DF logs over mavlink it would be really nice to download select sub parts of the log, by passing the autopilot a list of which messages you want. This would simplify the amount of data that would be needed to transfer for photo tagging, or most usage. It would allow for better usage of the MAVLink data over a RF link for situations when you don't want all the data.

My first thought was to use a bitmask to specify what to recieve, but @tridge pointed out that it wouldn't be to hard to instead pass the strings instead in for the messages and filter by that. (We should probably always send parameters by default and the FMT's for the messages you want)

WickedShell, 2015-11-16:
I don't know how I had a label clicked on that...

peterbarker, 2016-11-24:
Step 1: get a list of messages that *can* be downloaded.  We can assume that the FMT message is always supported, so downloading the entire log once filtering JUST to FMT messages would seem a reasonable way to do this.  It would be up to the GCS to pull apart this FMT-only dataflash file to extract the message IDs and message names to display to the user so they can select their fields.

Step 2: inform ArduPilot of a list of messages that are of interest.  Doing this with as few round-trips as possible is the interesting part.

```
          <enum name="MAV_LOG_FILT">
              <entry value="0" name="LOG_FILT_REMOVE">
                    <description>Stop filtering the logs</description>
               </entry>
              <entry value="1" name="LOG_FILT_FIELDS_CLEAR">
                    <description>Clear current list of fields before setting these ones</description>
               </entry>
          </enum
          <message id="xx" name="LOG_FILT_FIELDS">
               <description>Specifiy fields to filter to when downloading.</description>
               <field type="uint32_t" name="flags">flags; see ENUM_LOG_FILT</field>
               <field type="uint8_t[32]" name="field_ids">bitmask of fields IDs to download</field>
          </message>
```

Step 3: A download would then be initiated as normal.  The filter persists until reboot or the user modifies it.

rmackay9, 2016-11-24:
Another option which is quite different is to start saving the logs to companion computers (like the PH2's intel edison) and then retrieve the logs over wifi.

peterbarker, 2016-11-24:
@rmackay9's suggestion is how things work on Solo.  It's awesome.  But not everyone has  a companion computer :-)

WickedShell, 2016-11-24:
How would you imagine providing the list of which FMT's you want? I imagine a bitmask/numbering system would be really poor. When talking to @tridge about this originally he had the suggestion to just submit the character identifiers for the ones you want IE (GPS PARM CAM). I like this better then a bitmask of fields that the GCS has to know about/and keep updated. I think even if we sent these configs as a single message each (we could easily pack 5+ per config message) thats not a huge overhead as compared to the spinning through of the log and fetching the rest of the data.

Why LOG_FILT_REMOVE and LOG_FILT_FIELDS_CLEAR? If a GCS wants a non filtered log it should just call FIELDS_CLEAR and use the normal download. (and as long as the default is always to start with no filter, and if no filter then send everything).

Would the FMT only log just be a special log index that gets requested?

rmackay9, 2016-11-24:
The PH2 will be available with a read-to-use Intel Edison on the carrier board.  I just think that sometimes, you reach a point where throwing hardware at the problem is a much more efficient solution than a somewhat complex software workarounds to the basic problem.  The real issue here is that downloading logs over telemetry is slow.  So speeding that up with a companion computer and wifi really gets to the heart of the problem.

WickedShell, 2016-11-24:
@rmackay9 I think thats a really really poor general solution. An Edison has less storage available on board then the 4GB SD card that was provided with a Pixhawk. It also adds a lot of complexity to the system if you don't otherwise need a companion computer. (Linux does not like filling up it's storage, its one more then to check before the flight, requires a separate interface to manage afterwards/reset for the next flight, consumes power, adds weight).

peterbarker, 2016-11-24:
@WickedShell That's covered in my comment.  The GCS would first download a list of IDS/names (i.e. the FMT messages!).  They then send a bitmask back to the autpilot.

Why both LOG_FILT_REMOVE and LOG_FILT_CLEAR?  Because the empty set doesn't necessarily infer a full set should be inferred. We could use that magic instead, of course!

The FMT-only-log is just the full log filtered to FMT messages - it is just a very short DF log.

WickedShell, 2016-11-24:
Sorry I should have explained the question better: Is the FMT log actually going to exist or is just a hypothetical log that you can request and it will be generated on the fly?

In this case I'd say that no filter set should imply no filter which would be the full set, otherwise the process of setting it up is a bit more involved/unique. I'm fine if this isn't the case of course :)

peterbarker, 2016-11-25:
@WickedShell It will be generated on the fly.  

From a GUI UX experience, I would expect a user to nominate a log from a list.  They can click "Download Now" or "Download Filtered...".  If they click "Download Filtered" the GCS will fetch the log filtered to FMT mesages, and present a list of message types paired with checkboxes.  Theyt can then click a "Download" button to get the log.  I would expect the GCS to have FMT present in the list but the checkbox is selected and greyed out.

For MAVProxy:
 - `log formats 1` - get all of the formats in log 1
 - `log filt FMT,CAM,ATT,RATE,CRUD`
 - `log download 1`

I'm thinking it will throw a warning if you start to download while filtering.

DonLakeFlyer, 2016-11-25:
Couple requests:
* Make this mavlink generic support not ardupilot dialect
* Add a CRC to the download so you can verify. Current full log download mavlink support has no CRC which is kind of a bad thing.

peterbarker, 2016-11-25:
@DonLakeFlyer I was planning on using the existing log download messages for actually transferring the blocks.  Doing that means there's no scope for adding a CRC.

Trying not to create a new mechanism for downloading logs as much as possible - just adding the filtering people want.  This filtering will make CRCs harder, of course, as it can't be stored against the file, but would have to be generated as the filtered data was being sent to the client.

In terms of generic-mavlink - I'm not fussed which namespace it goes in, and the more input the merrier!

peterbarker, 2016-12-06:
After discussing with tridge, we'll be going with char[n] in the mavlink message instead of a bitmask: `CAM,ATT,GPS1`, for example.  

Late in the conversation the concept of decimation came in.

We intend in the future to move to a `CAM:1,ATT` format, where in this case `ATT` would come in at full rate but `CAM` would be decimated to 1Hz

peterbarker, 2016-12-15:
STABILIZE> log download 36
STABILIZE> Downloading log 36 as log36.bin
('Time has wrapped', 14675, 97211)
Finished downloading log36.bin (7458816 bytes 180 seconds, 41.4 kbyte/sec 944 retries)
STABILIZE> log filter add FMT,PARM
STABILIZE> log download 36
STABILIZE> Downloading log 36 as log36.bin
Finished downloading log36.bin (28944 bytes 16 seconds, 1.8 kbyte/sec 17 retries)

So it can filter at ~450kB/second

WickedShell, 2019-07-14:
Closing this out, as we never got the performance we were hoping to, and the PR that implemented this was sorta universally ignored.

closed 2019-07-14 by WickedShell