ArduPilot/ardupilot #3188: Plane: If unable to read the airspeed sensor do not use the value

WickedShell, 2015-11-16:
If the digital airspeed sensor can't be read for some reason for more then say 5 attempts? Then we should stop using the reported airspeed data.

magicrub, 2015-11-16:
If you mean it should revert back to internal airspeed value on data that is bad then this is a duplicate of https://github.com/diydrones/ardupilot/issues/1628

I guess there's two questions: data that is "bad" (ie, blocked tube) vs data that is stale due to comms/hardware problem.

Do those deserve separate issues?

WickedShell, 2015-11-16:
I had missed the other issue. I'm specifically thinking about a hardware problem here. I'm not sure if thats worth a seperate issue. It should be much easier however, simply ensure that the health bit is being set correctly, and if it's bad, then revert to the internal estimate.

If we want to track that on the other issue then we should just close this.

magicrub, 2015-11-16:
The trick is we can switch between, but we can't observe both at the same time. That's what's holding up the fix. no way to sanity check each other's sources.

closed 2015-11-16 by magicrub

WickedShell, 2015-11-16:
Right, this one doesn't need them to be both observed, this would simply be a swap when the hardware is not responding.

magicrub, 2015-11-16:
But the problem is you switch forever. If it's a temporary glitch then
you'll want the ability to switch back to the more accurate external
sensor. That said, right now it just lies to you and you crash because we
have no other sanity checking going on.