ArduPilot/ardupilot #3193: Telemetry - Add Light-telemetry (LTM) support

magicrub, 2015-11-17:
Issue request from https://github.com/marksev1 to add LTM for light-weight telemetry at low baud rates. Useful for antenna tracking and critical status at long range. For details, see
https://github.com/cleanflight/cleanflight/issues/36

marksev1, 2015-11-17:
Attaching ltm decoded file:

[ltm_decoded.txt](https://github.com/diydrones/ardupilot/files/36963/ltm_decoded.txt)

Here is a link to the raw stream:

https://drive.google.com/file/d/0B87FK33qoAT3UE9TcnpCcm55MTA/view?usp=sharing

Regarding antenna trackers: Ghettostation (its on github) supports it and amv-open360tracker is adding support for it almost as we speak. If you want to use a ground osd based on that telemetry you can use ltm-osd (max7456) to have a redundant ground osd if maybe the one in the plane fails because of loss of video range. Besides apm-planner and mission planner if you want another gcs to display telemetry you can with mwp (github, only for linux).

Taulabs supports it and cleanflight will support it in next release.
Otherwise I'm hoping you guys will like it as much as I and quite a few other people. Hopefully it will get in the codebase.

auturgy, 2020-05-06:
Closed by https://github.com/ArduPilot/ardupilot/pull/12158

closed 2020-05-06 by auturgy