ArduPilot/ardupilot #3208: Bebop Build fails

djnugent, 2015-11-20:
```
%% tuning.o
%% UserCode.o
%% libraries/AP_Common/AP_Common.o
%% libraries/AP_Common/c++.o
%% libraries/AP_Menu/AP_Menu.o
cc1plus: internal compiler error: Segmentation fault
no stack trace because unwind library not available
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.launchpad.net/gcc-linaro> for instructions.
make: *** [/tmp/ArduCopter.build/libraries/AP_Menu/AP_Menu.o] Error 1
```
Under investigation...

rmackay9, 2015-11-20:
i was able to "make bebop" on my Ubuntu 14.04 machine using master.  I guess you've followed these instructions?  http://dev.ardupilot.com/wiki/building-for-bebop-on-linux/

jberaud, 2015-11-20:
This compiler fault happens with older versions of g++. Could you tell me what arm-linux-gnueabihf-g++ --version outputs?

djnugent, 2015-11-20:
```
daniel@daniel-VirtualBox:~/ardupilot/ArduCopter$ arm-linux-gnueabihf-g++ --version
arm-linux-gnueabihf-g++ (crosstool-NG linaro-1.13.1-4.9-2014.07 - Linaro GCC 4.9-2014.07) 4.9.1 20140710 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jberaud, 2015-11-20:
Ok, strange. What is your Linux Distro?

djnugent, 2015-11-20:
Ubuntu 14.04

jberaud, 2015-11-20:
could you try with the official ubuntu compiler ?
sudo apt-get install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf

Then remove the other toolchain from your path and try again.

djnugent, 2015-11-20:
That complied. Thanks. I'll submit an issue to the docs

closed 2015-11-20 by djnugent