ArduPilot/ardupilot #3213: Bebop compass offsets too high

djnugent, 2015-11-21:
I have calibrated the bebop multiple times and my compass offsets are in the 700s. Not sure how to fix this on a closed system. @jberaud any suggestions?

rmackay9, 2015-11-21:
@djnugent, if you're using the MP with Copter-3.4-dev you'll find there are two methods for doing the compass calibration. One of them does all the logic in the ground station (the old method), then new version does it all on the controller.  There are buttons on the MP's compass calibration screen allowing it to be done using either way.  So.. I'd recommend trying them both.
Worst case, you might just try turning off the arming check.

jberaud, 2015-11-21:
@djnugent Are you on master or on a branch. Compass has been broken for a while so depending on where you are in history it might be broken. It works fine for me on master but I will try again on Monday and see if there are any issues.

djnugent, 2015-11-22:
@rmackay9  I have been doing groundstation based calibration with APM planner.  I was unable to get Mission planner to connect to bebop over UDP to activate on board calibration. Did you use any special settings to get mission planner to connect to the bebop?  

I am wary of using GPS modes until I am convinced the Bebop has robust manual control. Which brings me to a new question. Does a timeout on RC over UDP trigger RC failsafe?

@jberaud I am on master(assuming I uploaded the firmware properly).

rmackay9, 2015-11-23:
@djnugent, any loss of RC signals should trigger the regular radio failsafe so it should trigger if the bebop and mission planner lose contact.

jberaud, 2015-11-23:
@djnugent I still have to try and see if there are any current issues with compass calibration but do you think the values are inconsistent or does the compass rect normally apart from the high offsets?

djnugent, 2015-11-23:
@rmackay9 I just tested your claim and I think you are wrong. I had a look at the code and it appears the RC buffer only gets directly written to when a new UDP packet comes in. If no new UDP packet is received then the buffer remains untouched. So no timeout, therefore no RC failsafe. See here: https://github.com/diydrones/ardupilot/libraries/AP_HAL_Linux/RCInput_UDP.cpp#L51 

@jberaud they are consistently high. x: -130, y: 330, z: 560. Still unable to trigger on board calibration due to mission planner issues. I think I'l just fly it and hope for the best.

jberaud, 2015-11-23:
@djnugent I haven't looked into that but I will take a look about what triggers an RC failsafe and implement it for RC over UDP. If you have time you could implement it yourself :-)
Regarding the compass offsets, these are under 600 so what is the problem? I thought they were up 700.

djnugent, 2015-11-23:
@jberaud I will look into implementing it. Right now I just looking to get the copter up and flying safely. On average those are the compass offsets. Sometimes higher and lower. I was under the assumption that anything over 300 was considered high.

jberaud, 2015-11-23:
@djnugent honestly I don't know. I am more a low level dev and not at all an expert on drone calibration and stuff, I am still learning :-) 
So either the driver is broken or the calibration itself which is generic, or you offsets are high and it's normal

djnugent, 2015-11-23:
@jberaud You and me both. I think they are just normally high. What offsets have you seen in the past?

rmackay9, 2015-11-28:
@djnugent, maybe just turn off the pre-arm compass checks for now and see if it works?  We can adjust the acceptable offset threshold for the bebop if necessary.

djnugent, 2015-11-30:
I tried turning off prearm checks for an hour. Used a bunch of different combos and disabled them completely. Still couldn't arm. Cant even take off in alt hold anymore because ARMING_CHECK didn't appear to be working(not even 0). I got so frustrated I gave up. I'm going to do a rebuild and start from square one later this week.

rmackay9, 2015-12-01:
@djnugent, maybe you've got the fence enabled?  that causes the GPS checks to become required and they cannot be disabled with the ARMING_CHECK set to zero.

djnugent, 2015-12-01:
OK that would explain it for indoor flights.

When I was trying to do GPS flights I would get compass offsets too high and GPS vert velocity is 0.0 m/s. I can get rid of the compass offsets one but didn't feel comfortable disabling GPS prearm checks until I was aware of the effect of zero vert velocity on the ekf.

rmackay9, 2015-12-01:
@djnugent maybe try flying first in AltHold mode which won't require the GPS patches to pass before allowing take-off.

djnugent, 2015-12-01:
@rmackay9 Sounds good. I'll give that a try sometime this week. Thanks for the suggestion.

jberaud, 2016-01-11:
something that can also be done is disable the fan at startup (already done on bebop 2). On bebop 1, echo 0 > /sys/devices/platform/user_gpio/FAN/value before calibrating the magnetometer.

rmackay9, 2018-12-22:
I think we can close this because we've increased the default maximum compass offset and even have a parameter to allow the user to set the limit higher.

closed 2018-12-22 by rmackay9