ArduPilot/ardupilot #3216: List of PX4-only features to port to Linux

lucasdemarchi, 2015-11-23:
This is a list to keep track of the features available on PX4 that need to be ported to Linux.

Feature | Comment
---------- | ------------
Perf counters | missing command to get them from the flight stack
Arming switch | 
PWM input | See pwminput driver for PX4 - Need HW support and probably kernel driver
SW control of RCOutput enable/disable | 
Sensor auto-detection | see rc.APM, but should be done inside the flight stack
Runtime driver allocation | need INI parser + ifdef removal
IO mixed pass-thru in case the main FMU fails | need proper HW support, can be done by raspilot
Support for up to 14 PWM outputs | Most boards support it already - need clarification
Interface for time capture | Same as PWM input, need proper HW and kernel driver
Sanitize analog input | should use ADC rather than GPIO interface
UAVCAN | 
Relay support | 
Debug shell | *needed?*
OVERRIDE_CHAN | 
IRLock | 
RPM sensor | 
Power and ESC status writing to dataflash log |
Power status sent through mavlink |

lucasdemarchi, 2016-01-04:
Just updated the issue to turn it into a task list

EShamaev, 2018-02-17:
UAVCAN:
https://github.com/ArduPilot/ardupilot/pull/7466

auturgy, 2019-11-18:
@lucasdemarchi can you update or close?

closed 2019-11-19 by lucasdemarchi