ArduPilot/ardupilot #3225: Copter: Bug in loiter speed modification using CH6

thecodingfrog, 2015-11-24:
I've been able to change WP speed using CH6, but changing loiter speed doesn't work

rmackay9, 2015-11-28:
the wp-speed is separate from the loiter speed.  The wp-speed only affects the speed using missions.  
We don't currently support ch6 tuning of the loiter speed although it could be updated from the ground station's parameter list... but it won't take effect until the pilot switches out and then back into loiter.
I'm going to close this if that's ok but i'll still see comments posted here.

thecodingfrog, 2015-11-28:
So it is in the list but not available yet ?

Jean-Louis PERSAT

rmackay9, 2015-11-28:
@thecodingfrog, i'm afraid the loit-speed thing is not even in the list.

thecodingfrog, 2015-11-28:
Sure it is!

Jean-Louis PERSAT

thecodingfrog, 2016-05-28:
Any info since I've open this ?
Loiter speed is still available on CH6_OPT list but still doesn't change the loiter speed.
[image]

OXINARF, 2016-05-28:
That Loiter Speed label is indeed wrong. And no, this still hasn't been implemented. Anyway, we are changing tuning to a new library that Plane is already using, so this will probably be fixed then.

thecodingfrog, 2016-05-28:
That sound good

Jean-Louis PERSAT

thecodingfrog, 2016-10-24:
Hi,
Any news about this ?

OXINARF, 2016-10-24:
@thecodingfrog No, unfortunately no. We are in the final stages of releasing Copter-3.4 and this didn't make it. Hopefully we can get it in the next release.

thecodingfrog, 2016-10-24:
would be really interesting to have it :)
and beside this when we don't need any tuning from CH6 options would be also cool to be able to use CH6 for something else (parachute, autotune) like we do with CH7 et CH8

R-Lefebvre, 2016-10-26:
And at the same time, would be nice to be able to use Ch7 and 8 to do tuning different values,so you could tune 3 PID variables at the same time, for example.  I had done this almost 2 years ago but it didn't make it.

Naterater, 2019-01-08:
This issue is a bit old. Is there a possibility that this has been solved recently?

IamPete1, 2021-08-30:
duplicate of https://github.com/ArduPilot/ardupilot/issues/9782

closed 2021-08-30 by IamPete1