ArduPilot/ardupilot #3232: MAV_CMD_MISSION_START

khancyr, 2015-11-25:
Hi,

I try to use Mission start as described in wiki : http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_mission_start

When landed and a mission loaded with takeoff first, doing mission_start ,do nothing... got no response 

By looking in the code 
https://github.com/diydrones/ardupilot/blob/cbaa57c7be1fc7b373672225c63393f34f2831cd/ArduCopter/GCS_Mavlink.cpp#L1296

If I am landed in auto, I cannot arm , so we exit the case (and on mission planner we get no failed response) Is that the expected behavior? How to start an auto mission from land with auto takeoff ?

Thanks in advance!

rmackay9, 2015-11-28:
@khancyr, yes, this is by design as a safety measure.  So the only way to arm and take-off is either with the transmitter or using Guided mode.
I've renamed the title to clarify the request is to be able to arm and takeoff in Auto.  We need to be careful to implement this safely because we could end up with unexpected takeoffs otherwise.

khancyr, 2015-11-28:
@rmackay9 OK thanks ! I already do some test like that and it is working.
I found another way to auto-arm and takeoff with mission planner : by doing mission resume and start for first waypoint ( which is takeoff) , it switch to Guided, arm, and then takeoff but sometime it switch to Auto, sometime not ... I will make more test

R-Lefebvre, 2015-11-28:
Randy, I would propose we solve the safety problem by using something similar to the Motor Interlock method.  This makes take-offs a two-action system, so should be pretty secure.

So you would be able to arm directly into Auto, but nothing happens yet.  Then push the second button to enable the interlock, and it will begin the mission.  

We already have a double-action system, but the second stage is the pilot pushing up the throttle stick.  This is problematic as we want to get rid of throttle sticks in some cases.

The other double-action system, or this is more that two, is to arm in guided, do a takeoff, then switch to auto.  I guess it was considered that a guided mode takeoff is akin to the pilot pushing throttles up.

We do have to be careful here, I agree, as this is a risky thing.  It would be all too easy for users to have missions loaded, and forget that they did.  Then, they are using the copter some other day, and not realize they are in Auto.  They arm, thinking they're in Stabilize, but soon as they arm, it starts up and takes off.  

Another option, might be a MAVlink command called "Start Mission" or something like that. When the vehicle receives this message, Auto_Armed is set true.  Alternativesly, a MAVlink command Auto-Takeoff. Which sets Auto_Armed true, and performs a takeoff just like guided mode.

khancyr, 2015-11-28:
@rmackay9 @R-Lefebvre ,
You are right, that was what is was pointing out ! The code is in response to the mavlink message MAV_CMD_MISSION_START in GCS_Mavlink.cpp : 
````
case MAV_CMD_MISSION_START:
            if (copter.motors.armed() && copter.set_mode(AUTO)) {
                copter.set_auto_armed(true);
                if (copter.mission.state() != AP_Mission::MISSION_RUNNING) {
                    copter.mission.start_or_resume();
                }
                result = MAV_RESULT_ACCEPTED;
            }
            break;
````
So it should auto arm in response to MAV_CMD_MISSION_START (that is what is explain in wiki), no? But a agree about your concern about safety and it works pretty well currently!

lthall, 2018-01-12:
[Duplicate](https://github.com/ArduPilot/ardupilot/issues/4919)

OXINARF, 2018-01-12:
Closing this one in favor of #4919 which has more comments. Also takeoffs are allowed now, but arming still needs to be done in another mode first.

closed 2018-01-12 by OXINARF