ArduPilot/ardupilot #3239: Arducopter 3.2.1: Partial mavlink data to minimOSD on APM2.6

pwbecker, 2015-11-27:
I get some Mavlink data on the OSD (running MinimOSD-extra latest copter), such as flight mode, and voltage.  But I get no updates on Attitude or position, e.g. the GPS coords stay at 0.0, even after a GPS lock (tested via RTL).

If I connect a telemetry radio, all data is immediately displayed on the OSD.  If I connect via the USB, 'No Mavlink data' is displayed, but once the usb is disconnected, all data also appears on the OSD, including position data

I have only the Tx wire connected from the APM to the minimOSD (to allow an optional telemetry radio to be connected).  SERIAL1_BAID is 115, and SERIAL2_BAUD 57.

I have tried setting various combos of SR0 and SR1 parameters, to no avail.

pwbecker, 2015-11-27:
Could this be related to SR3/SR1 and the APM2 version?

mr-sneezy, 2015-11-27:
I've set up quite a few minimOSD's and have at times (on copters and planes) found similar issues (my most recent issue is the OSD and MissionPlanner altitudes being more than 100m different in flight). 
So what your seeing is not unusual to me at least, and I've had that exact issue as close as I can tell (and reported it).
What has helped me is reverting the minimOSD to the old 2.2 version and trying that out (after trying many SR1,2,3 settings). I've had more stability with the old code, and so have some of my friends...
(If you do this then remember change the character set to the one with 2.2 original as well)
I think this is now where that old version is found.
https://drive.google.com/folderview?id=0B4hykdYGOg4Td2dJNG40U0sxNDA&usp=sharing
Martin

rmackay9, 2015-11-28:
This is really more of a support issue so it probably doesn't belong in our github to-do list so I'm going to close this, sorry!.
The problem is likely the SRX_ parameters so you could try modifying the update rates through the mission planner.
http://copter.ardupilot.com/wiki/common-mission-planner-telemetry-logs/#setting_the_datarate

It could also be that the APM2.x simply doesn't have the CPU power to control the vehicle and send telemetry data.  If this is the case you'll likely see the telemetry works on the ground but not in the air.  You could also check to see if the data is displayed in your ground station or not.  If it's there then the problem is not in the flight code (aka arducopter, ardupilot, APM:Copter) and is more likely in the MinimOSD firmware.

closed 2015-11-28 by rmackay9

pwbecker, 2015-11-30:
I understand Mackay9.  Just some points: As reported, I did try all combinations of SR0 and SR1. Also, without the connection to a groundstation, it works 100% in the air and on the ground.

I logged an issue because there seemed a possibility that it could be to do with the APM2 backport incorrectly trying to use SR3 instead of SR1 to set up the serial port (in the absence of incoming Mavlink).

magicrub, 2015-11-30:
Have you tried minimOSD on a different mavlink channel or only snoop on the
GCS channel? When connected to another channel, and it still doesn't work,
try connecting the other data wire (minimOSD_tx -> apm_rx).

pwbecker, 2016-03-03:
**Please consider reopening this issue.**
I have looked into this further: The latest version I have of MinimOSD_Extra_Copter does NOT do a request_data_stream - the code is commented out.  APM does NOT send anything except heartbeats, until a request_data_stream is received.  It seems SR1_* are being ignored? (i.e. APM should send more than heartbeats, based on SR1_*)

Here is a log of a MVAlnk sniffer I wrote (first number is elapsed seconds)
==== Opening MAV 1...
==== Opening MAV 2...
0 > 0 (1, 1) > HEARTBEAT
1 > 0 (1, 1) > HEARTBEAT
...
26 > 0 (1, 1) > HEARTBEAT
27 > 0 (1, 1) > HEARTBEAT
27 < 0 (255, 190) < HEARTBEAT   <<<<<<<<<<<<<< turned on telemetry radio
27 < 66 (255, 190) < REQUEST_DATA_STREAM
27 < 66 (255, 190) < REQUEST_DATA_STREAM
27 > 42 (1, 1) > MISSION_CURRENT
27 > 35 (1, 1) > RC_CHANNELS_RAW
27 > 30 (1, 1) > ATTITUDE
27 > 74 (1, 1) > VFR_HUD
27 > 2 (1, 1) > SYSTEM_TIME
27 > 253 (1, 1) > STATUSTEXT
27 > 22 (1, 1) > PARAM_VALUE
...                                                                  <<<<<<<<<<<<<< turned off telem radio
33 > 22 (1, 1) > PARAM_VALUE
33 > 22 (1, 1) > PARAM_VALUE
34 > 22 (1, 1) > PARAM_VALUE
34 > 1 (1, 1) > SYS_STATUS
34 > 42 (1, 1) > MISSION_CURRENT
34 > 24 (1, 1) > GPS_RAW_INT
34 > 62 (1, 1) > NAV_CONTROLLER_OUTPUT
34 > 0 (1, 1) > HEARTBEAT
34 > 27 (1, 1) > RAW_IMU
34 > 29 (1, 1) > SCALED_PRESSURE
34 > 33 (1, 1) > GLOBAL_POSITION_INT
34 > 36 (1, 1) > SERVO_OUTPUT_RAW
34 > 35 (1, 1) > RC_CHANNELS_RAW
34 > 30 (1, 1) > ATTITUDE
...                                                     << All MAVlink keep flowing

pwbecker, 2016-03-04:
This problem doesn't occur in ArduPlane on the same hardware, so I found a difference in GCS_Mavlink which could be related

In ArduCopter/GSC_Mavlink.cpp: //                      Note difference in default values!
  AP_GROUPINFO("RAW_SENS", 0, GCS_MAVLINK, streamRates[0],  0),   
In ArduPlane/GSC_Mavlink.cpp:
  AP_GROUPINFO("RAW_SENS", 0, GCS_MAVLINK, streamRates[0],  1),