ArduPilot/ardupilot #3247: Feature Request: Mavlink Authentification and Securtiy (Authentifcation not Encryption ... later... )

netinetwalker, 2015-11-29:
I think Mavlink should get some security...

I think the current FightControll Hardware can't provide Full Encryption but it should be possible to provide basic securtiy for the current Systems.

Mavlink Security Levels?:
0. NO Authentifcation: Current situation
1. Basic Authentification for Control or Setting Frames (The Basic Step so noone can't hijack my Quad...)
2. Basic Authentification for All Frames
3. Full Encryiption like sMAVLink (on new Hardware?)

I think we can fast get to level 1 and 2 with the current Hardware.

All that we need are a token/salt/conter/hash based Frame Authentifcation!

Just my 2 cent

Best regards.

Neti Netwalker

Jman841, 2015-11-29:
+1 on this. I believe paparazzi already implements something similar, but, with the current fears of drones and people getting crazy doing silly things, a bit of protection from someone taking over your MAVLINK would be a huge improvement.

WickedShell, 2015-11-29:
This is the current target, that is being worked for that would provide
authentication:
https://docs.google.com/document/d/1ETle6qQRcaNWAmpG2wz0oOpFKSF_bcTmYMQvtTGI8ns/edit#heading=h.r1r08t7lr2pc

Provides authentication only, and is optional to use.

netinetwalker, 2015-12-05:
On issue #3275 more about mavlink filtering

But Authentifaction of the frames is the next step that should gone fast ...

WickedShell, 2016-06-06:
Signing is in master, and needs to be tested, but this issue isn't tracking anything fundamentally new. And the sMAVLink discussion should take place on the MAVLink issues/drone code server.

closed 2016-06-06 by WickedShell