ArduPilot/ardupilot #3249: Vagrant file SITL does not connect

hamishwillee, 2015-11-30:
As per [this discussion](https://groups.google.com/forum/#!topic/drones-discuss/3u6aK7__Z2A) the ardupilot vagrant file can build Ardupilot.elf but will sit in the MAV prompt forever waiting for a heartbeat.

I tried this building Copter both with the rsync and not-rsync options enabled.

```
vagrant@vagrant-ubuntu-trusty-32:/vagrant/ArduCopter$ sim_vehicle.sh -j 2
/vagrant/ArduCopter /vagrant/ArduCopter
Building sitl
Reading /vagrant/config.mk
%% libraries/AP_Common/AP_Common.o
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="084b6078"
%% libraries/AP_Common/c++.o
%% libraries/AP_Menu/AP_Menu.o
%% libraries/AP_Param/AP_Param.o
%% libraries/StorageManager/StorageManager.o
...
%% UserCode.o
Building /tmp/ArduCopter.build/ArduCopter.elf
%% ArduCopter.elf
Firmware is in ArduCopter.elf
/vagrant/ArduCopter
Starting up at CMAC : -35.363261,149.165230,584,353
 --model + --speedup=1vagrant/ArduCopter/ArduCopter.elf -S -I0 --home -35.363261,149.165230,584,353
Window access not found, logging to /tmp/ardupilot.log
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory:
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
```

hamishwillee, 2015-11-30:
I added a note to http://dev.ardupilot.com/wiki/setting-up-sitl-using-vagrant/ linking here to let users know that the vagrant build currently does not work.

maxtsap, 2015-12-05:
Try to run `$ sudo /vagrant/Tools/autotest/sim_vehicle.sh`, this worked for me.

hamishwillee, 2015-12-06:
Thanks @maxtsap - I've added this as a tip at the top of the page.

@gmorph Tridge asked me if I could ask you to look at this. ``/vagrant/Tools/autotest/`` is in the path on the Vagrant build, so you can run ``sim_vehicle.sh``  from /ArduCopter directory. However for some reason, while MAVProxy starts the system waits for a heartbeat and nothing else happens. It will work if you use sudo and start the build like this ``sudo /vagrant/Tools/autotest/sim_vehicle.sh``. All the files appear to be owned by vagrant group/user.

gmorph, 2015-12-07:
We shouldn't be running things as sudo.  If we are and something works then its more then likely a permission issue that needs to be sorted out.  I don't have Vagrant installed and have never used it.  If someone who is familiar with this could diagnose it that would be great.

hamishwillee, 2015-12-08:
FYI world, @andyp1per is planning to look into this issue.

hamishwillee, 2015-12-14:
@peterbarker created a fix for this https://github.com/diydrones/ardupilot/pull/3325 . Note that it only fixes vagrant - not native windows sitl or dronekit-sitl

commit 2c78b4f5 (2016-01-04): SITL: encourage bash to create a subshell when spawning ardupilot

This issue is evident in Vagrant if you don't forward X.

_fdm_input_step determined there was no parent and killed ArduPilot.

Closed #3249
closed 2016-01-04 by peterbarker