ArduPilot/ardupilot #3262: AP_InertialSensor: MPU6050 driver

mandel560, 2015-12-02:
I use mpu6050 as inertial sensor. But there is not mpu6050 driver. Can I use mpu9150 driver for mpu6050, or should I modify something for mpu6050?

rmackay9, 2015-12-03:
I'm not an expert (I didn't write either driver) but my understanding is that the mpu6050 is the same as the mpu6000 except that it doesn't have the SPI connection, it only has I2C.  So my guess is that you should use the MPU6000 driver.

I think the bebop also uses the mpu6050 by the way in case it's helpful to look at what driver they've used.  http://copter.ardupilot.com/wiki/parrot-bebop-autopilot/

I assume you're on a linux box and not a pixhawk compatible.  If it's a pixhawk compatible then the drivers will be in the px4firmware layer.

closed 2015-12-03 by rmackay9

mandel560, 2015-12-03:
I have tried MPU6000 driver for MPU6050, but it is always print as below. I don't know what's wrong with it.
```
bytes_read = 256, n_samples 21 > 3, dropping samples
bytes_read = 256, n_samples 21 > 3, dropping samples
bytes_read = 256, n_samples 21 > 3, dropping samples
bytes_read = 256, n_samples 21 > 3, dropping samples
bytes_read = 256, n_samples 21 > 3, dropping samples
```

rmackay9, 2015-12-03:
@mandel560, if this is a new board, it might be better to email drones-discuss@googlegroups.com.  If it's a new board then it's really more of a dev question so that'd be a better place for it.  This github issues list is more to capture known issues or enhancements.