ArduPilot/ardupilot #3269: Can't launch SITL

doronz, 2015-12-03:
I followed the instructions shown here: http://dev.ardupilot.com/wiki/copter-sitl-mavproxy-tutorial/

I then got a `Window access not found` error but still built successfully and launched map, console and mavproxy.

Couldn't figure out how to fix the above stated error, so I tried rebasing by `git pull origin master` on the `Ardupilot` folder. This gave me the latest code for ardupilot.

After running the same command again: `sim_vehicle.sh -j4 --map --console` I now get the following errors.

I could not find any resources online to help me fix this.

Suggestions?

```
%% ArduCopter.elf
/tmp/ArduCopter.build/libraries/DataFlash/LogFile.o: In function `DataFlash_Class::Log_Write_Parameters()':
/home/D/ardupilot/libraries/DataFlash/LogFile.cpp:718: undefined reference to `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
/home/D/ardupilot/libraries/DataFlash/LogFile.cpp:718:(.text$_ZN15DataFlash_Class20Log_Write_ParametersEv+0x3f): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
/home/D/ardupilot/libraries/DataFlash/LogFile.cpp:718: undefined reference to `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
/home/D/ardupilot/libraries/DataFlash/LogFile.cpp:718:(.text$_ZN15DataFlash_Class17Log_Write_SysInfoEPKc+0x5f): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
collect2: error: ld returned 1 exit status
C:/cygwin64/home/D/ardupilot/mk/board_native.mk:106: recipe for target '/tmp/ArduCopter.build/ArduCopter.elf' failed
make: *** [/tmp/ArduCopter.build/ArduCopter.elf] Error 1
/home/D/ardupilot/Tools/autotest/sim_vehicle.sh: Build failed
```

paiprashanth, 2015-12-06:
same problem here. Any progress on this issue,doronz?

squilter, 2015-12-07:
Works fine for me.  What operating system?

doronz, 2015-12-07:
Windows 10. Haven't figured it out yet.

paiprashanth, 2015-12-07:
I'm also using Windows 10.

squilter, 2015-12-07:
Has anybody gotten it to work in the past on Windows 10?

paiprashanth, 2015-12-07:
I faced the same issue while working on Windows 8.1 as well. Here's the snippet of error i'm getting

Building /tmp/ArduCopter.build/ArduCopter.elf
%% ArduCopter.elf
/tmp/ArduCopter.build/libraries/DataFlash/LogFile.o: In function `ZN15DataFlash_Class20Log_Write_ParametersEv':
/home/MSDFTFP/ardupilot/libraries/DataFlash/LogFile.cpp:718: undefined reference to `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
/home/MSDFTFP/ardupilot/libraries/DataFlash/LogFile.cpp:718: undefined reference to `DataFlash_Class::Log_Write_Parameter(AP_Param const*, AP_Param::ParamToken const&, ap_var_type)'
collect2: error: ld returned 1 exit status
C:/cygwin/home/MSDFTFP/ardupilot/mk/board_native.mk:106: recipe for target '/tmp/ArduCopter.build/ArduCopter.elf' failed
make: *** [/tmp/ArduCopter.build/ArduCopter.elf] Error 1
/home/MSDFTFP/ardupilot/Tools/autotest/sim_vehicle.sh: Build failed

Now, I'm wondering whether i'm going wrong with the SITL setup process. I'm following http://dev.ardupilot.com/wiki/sitl-native-on-windows/ , is there any additional code I have to modify other than the .bashrc file, where I add  "export PATH=$PATH:$HOME/ardupilot/Tools/autotest"

jrjbertram, 2015-12-07:
Seeing on Windows 7.1 as well.  Attempting to debug the error.. we'll see how far I get.

jrjbertram, 2015-12-07:
Looks like the problem was introduced somewhere between f55effa (fails) and 637b736 (builds).  Some of the commits between there don't compile for other reasons, so can't get any better than that.

So I suppose that if people want to work around it, they can:

git checkout 637b736
make clean; make sitl -j4

I'm not familiar with the codebase or checkins though, so can't say whether this build has other issues we'll encounter?  Maybe the dev instructions at http://dev.ardupilot.com/wiki/sitl-native-on-windows/ should be updated to build from a last-known-good tag or something like that.

Hope that helps folks until the root cause can be found and fixed.

- J.

brickta, 2015-12-08:
I'm pretty new to developing with arduplane, but it seems that the undefined reference is due to DataPath_Class not having a Log_Write_Parameters() with three arguments defined.  Adding "DataClass_Backend::" before the function clears that error (is defined in that class), but throws a "cannot call member function ... without object".  I guessing that one or more of the parameter values is not valid.  Something seem funny given that the token variable seems like it should be a pointer, but is declared as a struct variable (&token vs token).  The DataFlash_Class::Log_Write_Parameters() function structure looks similar to  AP_Param::show_all() function (that doesn't throw and error).

brickta, 2015-12-08:
This doesn't seem right, but changing line 718 allowed it to compile and make ArduXXX.elf (XXX=Plane in my case).  It clearly wanted the "DataClass_Backend" class function, but complained about not having an object to get the member function from.

backends[0]->Log_Write_Parameter(ap, token, type);	//added backends[0] object

Now my problem is that JSBSim --version doesn't report anything :(

paiprashanth, 2015-12-09:
Commenting line 718 lets the build to be complete works.  I'm not sure what parameter could be lost in the elf file. Until now, present file seems to work fine

jrjbertram, 2015-12-11:
Note I haven't had any issues running with version 637b736 for those who want to revert to that version.

rmackay9, 2015-12-12:
Personally, I always run SITL within a VM on my windows machine: http://dev.ardupilot.com/wiki/setting-up-sitl-on-windows/

Sorry, I know people want to run it natively but above is what I know works.

peterbarker, 2016-01-05:
Thanks for the problem report.

It is, indeed, a problem with a declared method not being defined.  Some compilers care, others (gcc!) don't.

I'm just about to make a PR cleaning this up.

commit ff8008d8 (2016-01-05): DataFlash: remove unused callbacks, unimplemented functions

Closes #3269
closed 2016-01-05 by peterbarker