ArduPilot/ardupilot #3270: compile copter3.3.2 error while HAL_CPU_CLASS < 150

19504643, 2015-12-03:
I have a flymaple board,the HAL_CPU_CLASS is 75 and I build it in linux following the step,but there comes the error Copter.h:415:5: error: 'AP_InertialNav_NavEKF' does not name a type
     AP_InertialNav_NavEKF inertial_nav;
I rollbacked to AC3.2.1,and it all went well,what's the matter?

magicrub, 2015-12-03:
Which commit were you trying to build ftom?
What command did you use to compile?
Have you tried the latest master?

19504643, 2015-12-03:
commit is 7f16e4...
command is "make flymaple" in Arducopter
not yet,I'll try
thanks

rmackay9, 2015-12-04:
So I think the issue is in the EKF it's got another set of #ifdef checks that only enables it for the CPU level above the flymaple.  We would just need to find and change those so that the EKF can be built.  Of course we don't know if the EKF will run on the flymaple, it may not because it's missing the floating point math co-processor.

19504643, 2015-12-04:
As I know,APM2.5 board runs the DCM,can copter3.3.2 support both DCM and EKF,and different boards can choose appropriate algorithm depending on their operation capability.

hiro2233, 2016-09-08:
@hiro2233

closed 2016-09-09 by 19504643