ArduPilot/ardupilot #3275: Mavlink SafeMode

Jace25, 2015-12-03:
Hello everybody,

Communicating via Mavlink is open and not secured. Each (evil) person with a GCS is able to listen in principle or even send commands to the Flightcontroller.

I think it makes sense to implement a mode that forbids making critical things while in flight. So the flight data and sensors could for example be available, but changing parameters is forbidden.

This mode could be automatically activated when the copter is armed.

People, who want to fly autonomously could disable this security feature (in the params).

magicrub, 2015-12-03:
This has been in the works for a very long time. Please follow this discussion, especially the links from Tridge to the DroneCode Technical Steering Committee which will be heading up this effort:
https://groups.google.com/forum/#!searchin/mavlink/dronecode/mavlink/u0UK_-wRVSI/5ke16sAmEwAJ

links:
https://docs.google.com/document/d/1ETle6qQRcaNWAmpG2wz0oOpFKSF_bcTmYMQvtTGI8ns/edit?pli=1#heading=h.r1r08t7lr2pc

https://docs.google.com/document/d/1XtbD0ORNkhZ8eKrsbSIZNLyg9sFRXMXbsR2mp37KbIg/edit?pli=1#heading=h.ovqxr52ozscu

Jace25, 2015-12-03:
Im not talking about the secure mavlink (sMavlink) implementations. I was talking about making ardupilot more secure about inflight changes. So in my opinion its not a direct mavlink related issue, but it is a direct ardupilot related issue because ardupilot has the opportunity to do inflight tuning, params changing and also flightmode changing. 

So maybe, beside the sMavlink diskussion (startet 2013?) it would be a nice feature to lock these features while ardupilot is armed or in action (eg. in flight)?

magicrub, 2015-12-03:
Ahh, gotcha. Not a secure thing, more like rejecting GCS commands to inhibit pirate GCSes from taking over.

Jace25, 2015-12-03:
Yes, exactly. My idea was to define a set of critical and non critical commands. In flight only the non critical commands are allowed (eg. read sensors), and all other commands are rejected. 

So the pirate gcs at worst is able to read sensor data (or other non crititcal data), but isnt able to crash the copter with setting stupid pid values or changing the flightmode.

WickedShell, 2015-12-03:
The packet signing will grant you all of that I don't see what you are
missing there? All the standard sensor data is all broadcast anyways?

magicrub, 2015-12-03:
He wants a "talk-only, do not listen" mode which is enabled by two
undetermined events such as arm/disarm or takeoff/land. It's a solution
that would work for auto-only flights in mavlink v1.x

WickedShell, 2015-12-03:
That's basically signing though is my point, (although it is true that it
doesn't acknowledge the arming states).

But if you're worried about rouge malicious attacks, then leaving yourself
in a unprotected listen state while disarmed is no better, as the malicious
vector could disable the functionality arm and do whatever they want.

It seems like a halfway solution in my mind, I'm not seeing the advantage?

magicrub, 2015-12-03:
It's an "available now" solution with a simple one of two line commit (plus
adding a new param). Is there another "available now" solution that is
better?

WickedShell, 2015-12-03:
My point is that it is far from a full coverage solution, and honestly
sounds more dangerous as now you can't disarm from the GCS or trigger a RTL.

magicrub, 2015-12-03:
not from a GCS, but you could from the RC input

Jace25, 2015-12-04:
I also think, this is a simple "available" workaround until Mavlink gets secured. And, while this workaround is controllable via Param, nobody is forced to use it. 

So it only gives you an optional security feature.

I'm a developer, but i never looked seriously inside the ardupilot code. But nevertheless i think, this feature could be implemented really simple (some kind of checking if the securtiy param is set when mavlink tries to change params).

WickedShell, 2015-12-04:
I'll but out as I appear to be the minority dissenting view. I just feel fairly strongly that something that offers no protection on the ground, and makes it easier for a malicious source to deny you any control of the aircraft when on the ground allowing them to do whatever they want is no better (or almost worse, as before you had a chance of reconnecting in flight with MAVLink). While I agree it shouldn't be to hard to implement I'd rather see the effort go into implementing the signing proposal that has been agreed upon, which offers a much more complete protection.

squilter, 2015-12-04:
Copter supports `MAV_CMD_DO_FLIGHTTERMINATION` in order to kill all motors while in flight.  This was implemented on purpose.  I don't understand why we would want to prevent somebody from changing PIDs from the ground to avoid crashing, when we specifically do allow them to crash with a mavlink command that essentially means "crash".

magicrub, 2015-12-04:
@WickedShell your input is most welcome! You're the minority out of 3,
that's not a very good sample set. More people are encouraged to chime in.

I don't think the security problem is as serious when on the ground (and
disarmed) since the signal range to a pirate GCS drops. It's after you get
airborne where your range, and visibility (visually and audibly) increases
and invites pirate GCSs. This is especially bad in BLOS missions.

magicrub, 2015-12-04:
The problem is not the ability to set PID/s and/or crash it. It's the
threat of taking over control and stealing property.

Scenario:
Take off with plane/copter and do an AUTO mission with 433/900MHz telemetry
available. While flying around someone else hears the drone, pulls out
Tower/MP and tries a couple NetIds. Once connected they upload new mission
to fly somewhere else, a landing location of their choosing. Once the drone
gets low on the horizon your antenna can no longer reach it and you can't
determine the landing location.

Proposed Solutions:
- total MAVLink makeover (v2.0) with security in mind. Release date and
configuration complexity is undetermined/unknown.
- new param that rejects MAVLink packet handling while armed (and while
flying?).

WickedShell, 2015-12-05:
The concern is that in you're BLOS scenario the RC is potentially
ineffective anyways without a LRS setup, and most malicious sources can
have high power equipment anyways. If you reject all MAVLink commands in
flight you can no longer initiate a large variety of safety actions (land
early, loiter, avoid known traffic, correct for a problem with the planned
mission that you just realized existed).

Stealing the vehicle is still possible while it's disarmed if the attacker
can see the MAVLink interface. (They can arm, disable other checks, turn
off RC overides etc). That's really why I don't like this, it significantly
limits you're ability to take a corrective safety action in flight, and
while it minimizes the attack surface a bit, and the exact vulnerability it
is protecting against is available on the ground, which can easily be
translated to taking over the craft anyways (at least for a copter where
taking off is easy). There is also a fair amount of predicting that can be
done for where someone will be flying a craft that allows this to be a real
problem, as you could be easily in range at the correct time to take
control, and why strictly worrying about the in flight aspect is a security
blanket in my opinion.

Mavlink signing should not be a hard proposal to implement, and once it's
present I'd strongly advocate that everything that this is proposing would
be obsoleted and should be removed. Tridge might have some input on a
realistic date for mavlink signing. I know some discussion early on was
talking about how signing could be done as an extension of MAVLink 1,
instead of strictly in 2, but I don't recall what the conclusion of that
was.

netinetwalker, 2015-12-05:
A basic Control of the mavlink interface is what is needed. Its like a firewall view on the protocal. 
Currently we have no auth of Ground Station or Quad. No signing on the frames. No filter.

Which feature should we get in the next months on the default firmware to prevent hijacking. Mavlink is used on many platforms and if i should build a drone defense system mavlink is one think i would look on...

Control on what mavlink can do with my aircraft is not a bad think.

Authentifcation should be the next step
Encryption the last step ...

magicrub, 2015-12-05:
Authentication work is in process, just an unknown release.

Jace25, 2016-01-09:
Just this:
http://www.shellntel.com/blog/2015/9/25/drone-code-execution

WickedShell, 2018-09-02:
I'm going to go ahead and close this. MAVLink signing is available and has been for quite awhile, securing your links is now your responsibility.

closed 2018-09-02 by WickedShell