ArduPilot/ardupilot #3280: AP 3.4: RSSI_CHANNEL can only be 5-8?

marcmerlin, 2015-12-04:
http://plane.ardupilot.com/wiki/arduplane-parameters/#receiver_rssi_channel_number_rssi_channel
says so, but I'm using all those channels.
My receiver is injecting RSSI on channel 16.
Can more channels be supported please?

If I set RSSI_CHANNEL to 8, I can see the RSSI value change in my OSD as I move my RC controller controls, so things works, but I just need it to read channel 16, not 8 :)

marcmerlin, 2015-12-04:
Mmmh, the issue is likely related to https://github.com/diydrones/ardupilot/issues/1040
First, I see that pixhawk does not support more than 14 input channels, but even changing my RSSI output to 14, does nothing because pixhawk seems to ignore any PPM past 12.

marcmerlin, 2015-12-04:
I've verified that channel 12 works for PWM RSSI input.
Can someone update the doc to say that channels 1-14 are supported (assuming that pixhawk can see all 14, and in my case it can only see 12)?

StewLG, 2015-12-05:
Marc:

As you've figured out, currently the code does not limit which channel you select for RSSI, so you can in fact use any channel that can be properly read by ArduPilot. 

I originally had it showing that any channel could be used for RSSI, but @Tridge specifically asked me to change it to 5-8. If I remember correctly he was trying to keep people away from the primary flight channels 1-4. I should have caught that it is really channel 5 or greater, and not just 5-8,

Thanks for the feedback.

marcmerlin, 2015-12-05:
@StewLG could it say 5-14 then or "5 and above" , or add etc... after the channel 8 example?
Currently it shows that it's limited to 5, and some things are actually limited to 8, like the backup CPU and mixing functions it can do on your things like your vtail.
Others like mode switch, or geofence switch are perfectly happy to be on channels over 8, but most people aren't really sure until it's spelled out to clear the confusion :)

marcmerlin, 2016-03-29:
Fixed by https://github.com/ArduPilot/ardupilot/commit/5bcd1b02eff5ae312f17dc0f6cb1adab5fca1420

closed 2016-03-29 by marcmerlin