ArduPilot/ardupilot #3284: Issue with RTL mode while using PX4 Octocopter Firmware

sahiljuneja, 2015-12-05:
Hello,

In reference to this post http://diydrones.com/profiles/blogs/3dr-pixhawk-powered-dji-s1000, I set up a DJI S1000 with a PX4. After having it set-up, I started it up, put it on AltHold and started the Auto-Tune process.

After the drone was tuned, I started it again, flew it a bit, and switched to RTL.

At this point, instead of the octocopter actually returning to the home position, it started to take a sharp turn and drift away, and started descending with quite a bit of forward tilt, and not towards the actual home position. 

I am not sure why this would happen. I am attaching the log file for any input and help on this. 

I used the Version 2 of the following firmware for the PX4 - http://firmware.diydrones.com/Copter/stable/PX4-octa/

This is the .bin log file. (I couldn't attach it here, even as ZIP, if you can't access it, please let me know.) http://uploadbin.net/e38ecb049b1d11e5a07b937cb31033d1/29.BIN

Any input on this? I am raising this as an issue, because I think this might be an issue with the latest arducopter code. The log file indicates an EKF Check Error which some people associate with the arducopter code, some with compass variance or vibration issues.

EricXiao1990, 2015-12-05:
Hi,
 I have got this with RTL, and It was very common because of bad GPS position. So I think it's not issue of arducopter code.

magicrub, 2015-12-05:
The logs will show that. Any copter experts available to verify that?

rmackay9, 2015-12-07:
This github issues list is really meant for confirmed bugs and enhancement requests. Support issues should be raised in the APM:Forum (http://ardupilot.com/forum/viewforum.php?f=111). I know that it's sometimes difficult to get help but at the same time we need to be strict about support items not appearing here or it will just turn into another support forum and we will lose track of the bugs and enhancements.

closed 2015-12-07 by rmackay9