ArduPilot/ardupilot #3285: convert all atof() -> strtof() possible environment configuration problem?

magicrub, 2015-12-05:
Contrary to common sense, atof() "Ascii to float" does not return a float, it returns a double. It's used in a few places where a float is expected so the correct strtof() "string to float" should replace it. However replacing it atof(str) with strtof(str, NULL), and including stdlib.h, the function is not found? Oddly, strtod() works.

This is when building in Eclipse with g++ 4.8.1 on the px4 env, win7, mingw32. When I built in Ubuntu with git 4.9.1 for SITL it works fine. Perhaps the problem is in arm-none-eabi-g++.exe?

WickedShell, 2015-12-05:
I wouldn't expect much of stdlib to be present when cross compiling for an
embedded arm, and past experience says what is there will have very
specific notes on what bits behave as you expect, and what has been omitted
for performance/memory/implementation reasons.

magicrub, 2015-12-05:
yup, makes sense. Embedded compilers are rarely fully implement the spec due to various resource constraints

magicrub, 2015-12-05:
Any suggestions on fixing this without casting?
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_LightWareSerial.cpp#L65

magicrub, 2015-12-07:
A few atof's were converted to strtof in https://github.com/diydrones/ardupilot/pull/3276 for SITL code but were excluded from the merge. I think it's safe to implement those but the AP_Rangefinder_LightWareSerial one linked above still does not have a good solution.

magicrub, 2016-02-21:
any ideas on this? I'm tempted to just cast to float but then we'll forget about it forever

OXINARF, 2019-07-12:
Has anyone tried strtof on ChibiOS? This is labelled as PX4 that we don't support anymore,

PR #18645 (2021-09-15): AP_Generator and AP_HAL_ChibiOS: Convert atof to strtof 
This converts the five remaining  `atof` in the flight code to `strtof`. Hopefully that is sufficient to close https://github.com/ArduPilot/ardupilot/issues/3285

We still have a few in Replay, SITL and sub-modules, but I wasn't going to bother with those.

Fixes #3285
closed 2021-09-15 by peterbarker