ArduPilot/ardupilot #3286: AC3.3.1 unexpected position change during auto LAND = Crash

imrj, 2015-12-05:
Copter was hovering normally at about 5m, send GCS LAND command and suddenly it veered off to the right rapidly then crashing into a parked vehicle (damage to both). 
GPS Sats and HDOP were fine, thou with N8M there is a dip in sats from 20 to 17, however this increase/decrease of sats from between 15-20 is normal for N8M and is seen many times thru this flight with no issue.
There is also a mysterious error 0-44 right before land command is issued, and coincidentally this crash happens. This copter has logged over 150 auto missions with never an issue like this and never had an error 0-44.
Please help....I dont believe this to be a GPS issue but am struggling to understand what precisely caused this and if there might be a software bug

[image]

magicrub, 2015-12-05:
Sorry about your loss. Can you post the log? It's the best way for the devs
to examine the behavior.

tabascoz, 2015-12-05:
The log is the image attached, just rename to .zip

tabascoz, 2015-12-05:
Right before the crash, line 217,4k , the roll /des-roll diverge. It could indicate a hw failure, however there is no reaction from px4 to compensate that... strange

WickedShell, 2015-12-06:
Unless I'm having a problem with mission planner's log review again I see no updates to you're GPS information at all shortly after you transition to land. And I don't see any new updates/changes until you enter loiter. If you never had a update here it makes a lot of sense why the copter drifted.

I will try and dig deeper on that log later this evening, but thats my first thought so far.

imrj, 2015-12-06:
hi all, i think i found the problem.....I run 2 GPS, and N8M and a 6M....for the duration of the flight the copter was tracking to GPS1 (N8M) but then suddenly (and for no apparent reason) you can clearly see right before the LAND command that the copter switches over to GPS2 (6M), am pretty sure this likely caused the drift in position...and the crash

however the question remains, why did AC switch GPSs when there is no apparent reason? the HDOP was much lower on GPS1 than GPS2 the entire time, my understanding is it should only switch GPS if the primary HDOP becomes higher than secondary, I have also heard conflicting reports about using or not using 2nd GPS.....I did thoroughly tested the GPS switch function a few months ago (by receiver activated switch, shutting off the primary GPS in flight) and yes there were similar changes in position as what happened during the crash land. 

Is running 2 GPS not recommended? are there issues with it such as this sudden and unexpected change in GPS? why did it switch GPSs when it wasnt suppose to?

[image]

rmackay9, 2015-12-07:
This github issues list is really meant for confirmed bugs and enhancement requests.  Support issues should be raised in the APM:Forum (http://ardupilot.com/forum/viewforum.php?f=111).  I know that it's sometimes difficult to get help but at the same time we need to be strict about support items not appearing here or it will just turn into another support forum and we will lose track of the bugs and enhancements.

In any case, I agree with WickedShell that it's most likely a mechanical problem, probably, motor 2 or 5.  This is consistent with the desired vs roll graph which shows the actual roll (green) suddenly diverging.  The vehicle rolls right.  Also if we look at the motor outputs we can see motors 2 and 5 climbing to their max while their opposites 2 and 6 go to their minimum (http://copter.ardupilot.com/wiki/connect-escs-and-motors/#hexa_octo_y6).

[image]
[image]

If this were caused by the GPS switch, then the desired and actual would have both spiked together.
By the way, we don't recommended dual GPS on copter.  The wiki warns of the issue re the sudden shift in position but we will make it more clear that we recommend sticking with just one GPS.

B

closed 2015-12-07 by rmackay9

mobimation, 2015-12-12:
Could it be a motor power glitch like in the early days with those ESC bullet cable connectors that lost conductivity ?