ArduPilot/ardupilot #3288: GPS detection timeout too short for ublox config to finish saving

jschall, 2015-12-05:
If you turn GPS_SAVE_CFG on with a ublox LEA-6H, " ublox " keeps getting printed out over USB and a fix is never obtained. The fix seems to be to increase the magic number in this line of code:

    if (tnow - timing[instance].last_message_time_ms > 1200) {

https://github.com/3drobotics/ardupilot-solo/blob/master/libraries/AP_GPS/AP_GPS.cpp#L332

jschall, 2015-12-05:
Also happens even if GPS_TYPE is set to 2 (ublox)

WickedShell, 2015-12-06:
Hmm... Yeah that makes sense I guess. I didn't see the problem before but I had only tested with a M8T which saved the configuration much faster. Hopefully I can reproduce this with the LEA-6H I have (It has a timing firmware flashed).

Sadly just raising this would address this issue and probably make a couple of other issues seem to do better ie #3126 

I'm quite happy to raise the limit, from a users perspective I'm actually comfortable with transitioning the GPS driver to load a GPS driver once it auto detects the type, then never unload it, just send the config strings if we ever lose it/re run the init procedure. The only downside to this approach is that you couldn't unplug/swap gps types without power cycling, however this is an extremely rare situation and almost exclusively done by devs, and I think on swapping types of GPS power cycling is an accetable solution.

The reason it still happens with setting the GPS_TYPE is that GPS_TYPE is really only a hint to the autodetect algorithm as to what kinds of things it can detect, and is never used to bypass the algorithm.

Thoughts on not  allowing the gps driver to be unloaded once we auto detected successfully? @tridge @rmackay9 @jschall

jschall, 2015-12-06:
Also I notice that the GPS frontend itself seems to have serial-specific
code in it (baud rate configuration). Doesn't that really belong in the
drivers? The frontend shouldn't care about the interface.

WickedShell, 2015-12-06:
The front end typically has to cycle through all the baud rates to run the auto detect. Otherwise there isn't a good way to run with auto baud (or grab a GPS that's on the wrong baud). IE we jump all ublox to one of 2 bauds, but that is not compatible with detecting for a different driver at a different baud rate. The ublox driver handles this on the conclusion that the front end has always found the correct baud for it.

WickedShell, 2016-02-08:
Closing this, as the solution @tridge is happier with is to simply warn that this is not expected to work on a 6 series GPS if attempted.

If people continue to have problems with this/we need to readdress it we can either reintroduce the concept that the GPS will be unresponsive for a period of time, or add checks to not run the save logic if we detect a u-blox 6 or below.

closed 2016-02-08 by WickedShell