ArduPilot/ardupilot #3291: Cannot Arm pixhawk without turning off 16/INS in ARMING_CHECK

marcmerlin, 2015-12-06:
As explained in http://ardupilot.com/forum/viewtopic.php?f=113&t=13999

AP 3.4, pixhawk, 
I have HDOP at 1.07
GPS FIX: 4.0
GPS sats: 9
Arming gives prearm: gps speed error 1.0

I had to wait a long time (minutes) before getting
HDOP 0.86
Fix: 4.0
Sats: 11
and finally I was able to arm
Strangely if I power cycle, the next time it just arms without problems with HDOP 1.32.

More tries, 
DEBUG 2015-10-30T08:54:46.593 APM: Text Message rx'd "PreArm: GPS horiz error 5.4" 
DEBUG 2015-10-30T08:58:00.927 APM: Text Message rx'd "PreArm: GPS speed error 1.5" 
DEBUG 2015-10-30T08:54:46.593 APM: Text Message rx'd "PreArm: GPS horiz error 5.4" 
DEBUG 2015-10-30T08:58:00.927 APM: Text Message rx'd "PreArm: GPS speed error 1.5" 

I've now been flying with ARM_CHECK 2030 
Even assuming that my gyro is mis-calibrated (I did calibrate it not that long ago), why am I getting GPS arm errors.

Dataflash:
http://ardupilot.com/forum/download/file.php?id=4653

rmackay9, 2015-12-07:
This github issues list is really meant for confirmed bugs and enhancement requests. Support issues should be raised in the APM:Forum (http://ardupilot.com/forum/viewforum.php?f=111). Which is what you've done so I'll close it here.
I know that it's sometimes difficult to get help but at the same time we need to be strict about support items not appearing here or it will just turn into another support forum and we will lose track of the bugs and enhancements.

closed 2015-12-07 by rmackay9

marcmerlin, 2015-12-07:
Fair enough, it's not technically a confirmed bug, but would you agree that
getting GPS speed errors linked to the INS setting sounds fishy?
If you think it's possible for this to happen without a bug in the code,
let's keep this closed as you said.