ArduPilot/ardupilot #3294: 3.3.1 Hexacopter motor redundancy

roque-canales, 2015-12-07:
Hello,
I'm tring to tune my hexacopter to handle motor failure.
For that I set low YAW IMAX and low YAW HEADROOM.
I make few tests taking off with one disconnected motor, and I'm achieving good stability.
If it is empty:
    It flies as if all the 6 engines were in working order.
If it is loaded:
   It flies correctly, but it rotatesslowly on its yaw axis.
   He continues to lock its position well.
   RTL works well and it's possible to control easily via the simple mode.

Now the tricky case it's to handle failure in flight.
For that I found modificated AC3.2 compiled by Randy with CH7 Option to stop one motor or limit power on one motor in flight

AC3.3 it's different, due to stability patch since added (haw headroom...), also it will be interesting to be able to test motor failure in flight setting witch motor we want to test (from a to f).

There's any one interested and able to share AC3.3.x compiled version with motor failure ch7 opt capability?

tabascoz, 2016-12-18:
Hello, I would like to inform a crash with one real motor failure. [http://discuss.ardupilot.org/t/hexa-crash-one-motor-failure/13576?u=tabascoz](url) Willing to test and help on this matter!

OXINARF, 2016-12-19:
Let's track this in #838

closed 2016-12-19 by OXINARF