ArduPilot/ardupilot #3297: Missing ./waf

khancyr, 2015-12-07:
Hi,

@guludo @lucasdemarchi 

Since the commit of the new build system,  waf binary is missing ... Is that normal ?
On a fresh repo, we got : 
./waf: No such file or directory
and on travis log you can find too: 
Tools/scripts/build_all_travis.sh: line 37: ./waf: No such file or directory

lucasdemarchi, 2015-12-07:
As we talked to @tridge we decided to include a copy the waf repository as a submodule. I think you are having problem due to having a wrong alias for waf

khancyr, 2015-12-07:
so alias should be : 
alias waf="$PWD/modules/waf/waf-light" ? 
in that case it works (at least until gbenchmark...)

If you check recent travis build you will see the same issue

lucasdemarchi, 2015-12-07:
@khancyr yep... the travis script needs to be fixed as well. @guludo, could you work on it?

guludo, 2015-12-07:
Oh yeah! Thanks @lucasdemarchi and @khancyr! I'll fix that right now.

closed 2015-12-08 by khancyr